OpenBoardView/OpenBoardView #130: not compatible with osx 10.10.5

Wolfcare, 2018-06-14:
Hi Paul, it's says compatible with osx 10.9 or higher but i get a forbidden sign on the app.

piernov, 2018-07-03:
Indeed, probably using the default target of the Travis VM (which is 10.11 now IIRC).
In the meantime, I built the master branch manually on Yosemite so it should work for you at least. DMG attached.

[OpenBoardView-R7.4-Darwin.dmg.zip](https://github.com/OpenBoardView/OpenBoardView/files/2160072/OpenBoardView-R7.4-Darwin.dmg.zip)

Wolfcare, 2018-07-03:
while the app doesn't have a prohibited sign it crashes on launch

piernov, 2018-07-03:
What's your hardware configuration? (esp. GPU)
Can you try to run it in the terminal with the `-r 1` and `-r 2` parameters?
With openboardview in the Applications folder, start Terminal and type:
```
/Applications/openboardview.app/Contents/MacOS/openboardview -r 1
```
and
```
/Applications/openboardview.app/Contents/MacOS/openboardview -r 2
```

Alternatively you can edit the `~/Library/Application Support/OpenBoardView/obv.conf` file and change the value of the `renderer=` line to `1` or `2`.

PS: I don't have real Apple hardware so this version has been tried in a virtual machine only, previous ones were tried on a hackintosh.

pvtinflex, 2018-07-07:
@piernov  given that Travis builds with a default 10.12 / Xcode 8.3.3 by default, would that be influencing this?  I believe we couldn't push it back further due to issues with ImGui?

Wolfcare, 2018-07-07:
I went into the package and was able to launch the app via terminal.
I’m probably going to upgrade to Sierra on all machines in the office shortly, So no worry.

Thanks for replying and effort.

Mike

_____________________________
Wolfcare @ Aqua Carre
Entrance C, 3rd floor, room 313  
Ritter str 26, 10969, Berlin 
Telephone: +49 15754386391
info@wolfcare.de
www.wolfcare.de
https://www.facebook.com/wolfapplecare

piernov, 2018-07-20:
@pvtinflex probably need to set the `CMAKE_OSX_DEPLOYMENT_TARGET` in the top-level CMakeLists.txt to 10.9 or something like that. OBV itself builds and works on 10.9, and 10.7 with small tweaks to `osx.mm`.

inflex, 2018-07-20:
I'll have a fiddle around later and see how far we can take it back. I'll also text on macOS High Sierra to ensure it still works with current releases.

piernov, 2018-07-20:
@inflex Last time I tried on my hackintosh with High Sierra it was fine.
At least the DMG I built above should work with 10.10 to 10.14.
I still did not have the time to switch to 10.14 though.

Wolfcare, 2018-07-20:
i have tried 10.10 on multiple machines no sucess has always a circle with
a strike aka no access but no matter