OpenBoardView/OpenBoardView #131: MiniGW Runtime Assertion - Failed

orrshemesh, 2018-06-23:
While trying to open a known good .brd file, the following error is displayed:
Title: MiniGW Runtime Assertion
Content:
Program: ___path___\openboardview.exe
File:
/home/travis/build/inflex/OpenBoardView/src/openboardview/FileFormats/BRDFile.cpp, Line 168
Expression: format.size() < num_format

If trying to open with ErrorMessages version of openboardview, the following message is displayed:
[image]

pvtinflex, 2018-06-23:
Can you attach the board file here as a zip?

Looks like it has a parameter in it that's out of range or causing a hiccup with the parsing.

orrshemesh, 2018-06-23:
is there a way to send it to only one person? i wish not to upload it publicly

pvtinflex, 2018-06-23:
You can email it to me at pldaniels@gmail.com

orrshemesh, 2018-06-23:
sent, thank you.

lalit11486, 2019-05-04:
pls guide for this error 
[Doc1.docx](https://github.com/OpenBoardView/OpenBoardView/files/3144123/Doc1.docx)

MattSturgeon, 2019-05-12:
@lalit11486 that is a different error. Please open a [new](https://github.com/OpenBoardView/OpenBoardView/issues/new) issue.

piernov, 2019-07-26:
Any news on this one? It is however always better to attach the file to the issue so that everyone can try to take a look at it.

(PS: cleaned irrelevant and unrelated comments)

closed 2025-08-20 by piernov