OpenBoardView/OpenBoardView #132: Error on opening Asus FZ files

anshadedavana, 2018-06-28:
I have a bunch of Asus board views. Most of them are in FZ format and some are in ASC format. While OpenBoardView can open and view ASC files, it is having trouble opening the FZ files. At the same time Asus 'PCB Repair Tool' program can open them so the files are not corrupted.
[image]

[boardview _ X450LC_2.0.zip](https://github.com/OpenBoardView/OpenBoardView/files/2143745/boardview._.X450LC_2.0.zip)

A sample FZ file is attached.

anshadedavana, 2018-06-28:
I am closing this issue as its a duplicate. If anyone know how to fix this key thing, please email me at anshadedavana@gmail.com

closed 2018-06-28 by anshadedavana

pvtinflex, 2018-06-28:
Get the key from the IRC chat channel  irc.freenode.net #openboardview

anshadedavana, 2018-06-28:
Hi

Got the key via IRC link. Many thanks for helping me.  Amazing software.
Loving it.