OpenBoardView/OpenBoardView #133: Unable to open file with quotes in file path

Polymer15, 2018-06-28:
Was trying to open file:

`\Dropbox\Boardviews\Macbook Pro 15" 2017\boardview.brd`

Getting error:
`Error opening the file: "\Dropbox\Boardviews\Macbook Pro 15? 2017\boardview.brd"`

Changed the filepath to not include quotation, issue was resolved. The file opener seems to have issues parsing filenames with " in the path.

MattSturgeon, 2018-07-10:
Could you confirm which methods you have tried to opening this file with?

e.g. open obv and then use the open file menu; open a command line and include the filename in the obv command; double-clicking a BRD file in _X_ operating system's file browser

It is possible that some of these scenarios behave differently, depending on where the bug is in the code.

Polymer15, 2018-07-10:
Hey Matt. Personally, it was when opening via the "File > Open" GUI.
However I have not tried the command line interface

sasodoma, 2018-12-01:
I'm having a similar problem that I think is caused by the same bug. My file path doesn't contain any quotes but it does contain a special character "š". Changing the path in my case is a bit trickier as it's a part of my username. As for the opening method, the error happens when using File > Open, the other two ways you mentioned don't work at all - when using the command line and appending the filename, obv opens but doesn't even try to load the board. If I double click on the brd file I get a Windows "open with" dialog. Selecting obv here also makes it open but not load the board.

piernov, 2018-12-19:
Mostly related to `file.open()` in https://github.com/OpenBoardView/OpenBoardView/blob/master/src/openboardview/utils.cpp#L20
`stat()` in https://github.com/OpenBoardView/OpenBoardView/blob/master/src/openboardview/main_opengl.cpp#L333 needs to be fixed as well.
Issue is that those functions only support ASCII character set on Windows. To support unicode character set Windows requires that functions with wide chars be used. `ifstream::open` doesn't support wide chars on MinGW (a non-standard overload is present in MSVC).
See https://stackoverflow.com/questions/821873/how-to-open-an-stdfstream-ofstream-or-ifstream-with-a-unicode-filename

One solution is to implement platform-specific functions for those features. Windows implementation could use WinAPI while other platforms could use existing code.
A faster way to fix `file_as_buffer()` is to use the non-standard GCC function `__gnu_cxx::stdio_filebuf<char> fb (fd, std::ios::in);` on Windows with fd beeing a file descriptor that can be opened with a wide char name on with `_wfopen()`. Would make it only compatible with GCC most likely.

piernov, 2021-06-21:
Opening files with non-ASCII characters on Windows should be fixed in the latest development release.
Double quotes in filenames are not supported on Windows so it's a no-fix for Windows, but they seem to be working fine on Linux.

closed 2021-06-21 by piernov