OpenBoardView/OpenBoardView #135: New Format  - Samsung 'CAD' file support

anshadedavana, 2018-07-08:
Samsung laptop boardview files usually came in the format with extension '.cad'.  I am not a programmer and hence has no idea about the internal structure of the file. However Remax's BoardViewer 1.2 app can open it so i assume the format is not encrypted.

[Samsung boardview.zip](https://github.com/OpenBoardView/OpenBoardView/files/2173472/Samsung.boardview.zip)

pvtinflex, 2018-07-08:
Yes, this is another plain-text format, importer can be made for it, likewise the other format too that you posted up.

inflex, 2018-07-21:
Bit hard to decide what to put in for the format testing. There's no real single key ID tag, but some of the attributes are moderately unique, such as the presence of;
```
P_ADDP 
```
or 

```
###Panel Added Part Information
```

Not sure how consistently their existence is.  Perhaps combine with the .cad extension ?