OpenBoardView/OpenBoardView #138: Please provide an AppImage for Linux

probonopd, 2018-07-20:
Providing an [AppImage](http://appimage.org/) would have, among others, these advantages:
- Applications packaged as an AppImage can run on many distributions (including Ubuntu, Fedora, openSUSE, CentOS, elementaryOS, Linux Mint, and others)
- One app = one file = super simple for users: just download one AppImage file, [make it executable](http://discourse.appimage.org/t/how-to-make-an-appimage-executable/80), and run
- No unpacking or installation necessary
- No root needed
- No system libraries changed
- Works out of the box, no installation of runtimes needed
- Optional desktop integration with `appimaged`
- Optional binary delta updates, e.g., for continuous builds (only download the binary diff) using AppImageUpdate
- Can optionally GPG2-sign your AppImages (inside the file)
- Works on Live ISOs
- Can use the same AppImages when dual-booting multiple distributions
- Can be listed in the [AppImageHub](https://appimage.github.io/apps) central directory of available AppImages
- Can double as a self-extracting compressed archive with the `--appimage-extract` parameter

[Here is an overview](https://appimage.github.io/apps) of projects that are already distributing upstream-provided, official AppImages.

If you have questions, AppImage developers are on #AppImage on irc.freenode.net.

piernov, 2018-07-20:
I'm personally not fond of AppImage (if at all). OpenBoardView itself is pretty portable and easy to build on most systems and I don't feel like the other perks will be of any use for us.

However we could consider adding support to our (Travis) continuous integration if you make a pull request for it.

inflex, 2018-07-20:
As a very long time linux user, I've seen this new proliferation of snap, appimage, Flatpak ... and maybe docker; seems more like the classic XKCD standards proliferation.  Instead of now just having .tar.gz/deb/rpm, we'll now have .tar.gz/deb/rpm/appimage then flatpak, then...

To the best of my knowledge, the packages built from the Travis-CI system should already have static build dependencies.

Maybe if someone wants to sort out the build.sh/CMakefile/.travis.yml for this and offer it as a PR, else I'm not sure if any of the existing devs have the time to do this ?

closed 2018-08-17 by piernov

wolfv, 2018-10-01:
I was actually right-away looking for a version that's flatpak'd. If you're open to the idea, I could work on getting OpenBoardView on FlatHub, which is a convenient, cross-distro way of distributing free software.

https://flathub.org/home

piernov, 2018-10-01:
I honestly don't know what it involves. We would most likely be open to contributions to port it to another packaging system as long as it's well integrated with the current build system (even with the CI if possible) and doesn't involve extra maintenance.
About submission on Flathub, reading the https://github.com/flathub/flathub/wiki/App-Submission documentation, especially paragraph `Someone else has put my app on Flathub—what do I do?` tells me that the developers should keep control, so the submission would have to been made as the "OpenBoardView organization" (as in Github organization). I don't know what's the process.

probonopd, 2019-07-26:
I have done the work some time ago; please review and eventually merge #160 (or tell me what needs to be changed). Thanks for your kind consideration.