OpenBoardView/OpenBoardView #140: (a bit offtopic) converting/measuring Apple brd files

reijin90, 2018-08-10:
Hi!

I want to measure some distances, or better yet convert the Apple brd files I have to use them in a CAD program (e.g. as an SVG). The board size must not be changed of course, so the measurements must stay true. Do you guys have any idea how I can figure out what filetype it is or any other program where I can measure distances accurately or convert it?

Thanks so much for any input, it's been really hard for me to find something useful.

Joebeazelman, 2021-05-29:
Without knowing the specifics of what you're trying to do, I will try and give a solution that works very well. I don't believe OpenBoardView supports file exporting or printing. OpenBoardView is one of the few programs that can open Apple PCB board files.

It might sound far fetched, but can get fairly accurate measurements by scaling the PCB to a even ratio, such as 100% or 50% and taking a screenshot of it. The dimensions should be spot on especially is you use a higher scale factor such as 200% or 400%. The issue here is that you'll either need a large enough monitor to fit the entire board on the screen or cut it into pieces and stitch it together in a bitmap editor such as Photoshop. If you don't believe it, hold up a ruler to the screen.

You can import the bitmap into a high quality vector editor like illustrator or Inkscape. Both of these programs have a trace feature that traces the bitmaps to form vector polygons. Given the simple polygon of the PCB and components, it should produce high quality vector shapes. You can also manually trace the polygons using the tools menu, but it can be a bit tedious. 

If you want the best results, you can try to turn off unnecessary graphics and text such as component names, wires, etc. before taking a screenshot. The simpler the geometry, the more efficient the resulting output. If Open Board View supports it, you should screenshot each PCB layer in isolation and import them into separate layers in your vector editor. This will simplify the geometries. Once you're done with your tracing, scale the entire vector image to 100% and export it to any format compatible with CAD program. Scaling is simple. If it's 50% scale it by 2 to get 100% scale, or if it's 400% scale it by .25.

Hope this is helpful.

reijin90, 2021-05-29:
Unfortunately, the project I worked on did not pan out the way I hoped. So I'm not going to make use of your answer. 
Regardless of that, I wanted to thank you for your thoughtful answer and the time you took to send it. At the very least, someone else can hopefully make use of it. 

Cheers! :)

closed 2021-05-29 by reijin90

Joebeazelman, 2021-06-04:
@reijin90  If you don't mind me asking, what was the project you were working on?  It's sounds very interesting.

reijin90, 2021-06-04:
Sure!
I wanted to create a unified Macbook board case that would have the correct form factor and mounting to allow fitting Macbook boards inside. I wanted to create a cheap solution that would allow making "mac minis" out of Macbooks which displays, mouse, etc are broken. However, this is a completely new field for me and I didn't get funding to build this. 
I still like the idea a lot, but it requires skills I do not yet have and now don't really have the time to develop.

Joebeazelman, 2021-06-04:
Hmmm. Interesting!  Seems like a great way of keeping MacBook Pros from ending in the landfill. These machines are far too valuable to chuck in the trash.  I wish you luck in your endeavor!