OpenBoardView/OpenBoardView #148: build error: imgui.h: No such file or directory

Apteryks, 2018-12-03:
It seems the file src/openboardview/Renderers/imgui_impl_sdl.cpp should include its imgui.h header as "imgui/imgui.h" (as done everywhere else), otherwise it isn't known:

````
src/openboardview/Renderers/imgui_impl_sdl.cpp:10:19: fatal error: imgui.h: No such file or directory
````

MattSturgeon, 2018-12-08:
How are you building obv? iirc the way CMake works, all includes should work fine because we use [`include_directories()`](https://github.com/OpenBoardView/OpenBoardView/blob/1caeb67a9f1e8bfa6fbf8496312fa1731caf335e/src/openboardview/CMakeLists.txt#L47-L58)

piernov, 2019-07-26:
Indeed, there's no reason for this not to work. Please re-open if there's an actual issue with CMake.

closed 2019-07-26 by piernov