OpenBoardView/OpenBoardView #149: OpenBoardView cannot be built with latest upstream ImGui v1.66

Apteryks, 2018-12-06:
When attempting to use the upstream ImGui library to see if I could escape an assert error in the ImGui library bundled with OpenBoardView, I found that upstream has now drifted a little bit and cannot be used with OpenBoardView. Below are the errors thrown by GCC:

````
/tmp/guix-build-openboardview-7.3.drv-0/openboardview-7.3-checkout/src/openboardview/BoardView.cpp: In member function ‘void BoardView::ThemeSetStyle(const char*)’:
/tmp/guix-build-openboardview-7.3.drv-0/openboardview-7.3-checkout/src/openboardview/BoardView.cpp:90:16: error: ‘ImGuiCol_ComboBg’ was not declared in this scope
   style.Colors[ImGuiCol_ComboBg]              = ImVec4(0.20f, 0.20f, 0.20f, 0.99f);
                ^
/tmp/guix-build-openboardview-7.3.drv-0/openboardview-7.3-checkout/src/openboardview/BoardView.cpp:106:16: error: ‘ImGuiCol_CloseButton’ was not declared in this scope
   style.Colors[ImGuiCol_CloseButton]          = ImVec4(0.50f, 0.50f, 0.90f, 0.50f);
                ^
/tmp/guix-build-openboardview-7.3.drv-0/openboardview-7.3-checkout/src/openboardview/BoardView.cpp:107:16: error: ‘ImGuiCol_CloseButtonHovered’ was not declared in this scope
   style.Colors[ImGuiCol_CloseButtonHovered]   = ImVec4(0.70f, 0.70f, 0.90f, 0.60f);
                ^
/tmp/guix-build-openboardview-7.3.drv-0/openboardview-7.3-checkout/src/openboardview/BoardView.cpp:108:16: error: ‘ImGuiCol_CloseButtonActive’ was not declared in this scope
   style.Colors[ImGuiCol_CloseButtonActive]    = ImVec4(0.70f, 0.70f, 0.70f, 1.00f);
                ^
/tmp/guix-build-openboardview-7.3.drv-0/openboardview-7.3-checkout/src/openboardview/BoardView.cpp:179:16: error: ‘ImGuiCol_ComboBg’ was not declared in this scope
   style.Colors[ImGuiCol_ComboBg]              = ImVec4(0.86f, 0.86f, 0.86f, 0.99f);
                ^
/tmp/guix-build-openboardview-7.3.drv-0/openboardview-7.3-checkout/src/openboardview/BoardView.cpp:195:16: error: ‘ImGuiCol_CloseButton’ was not declared in this scope
   style.Colors[ImGuiCol_CloseButton]          = ImVec4(0.59f, 0.59f, 0.59f, 0.50f);
                ^
/tmp/guix-build-openboardview-7.3.drv-0/openboardview-7.3-checkout/src/openboardview/BoardView.cpp:196:16: error: ‘ImGuiCol_CloseButtonHovered’ was not declared in this scope
   style.Colors[ImGuiCol_CloseButtonHovered]   = ImVec4(0.98f, 0.39f, 0.36f, 1.00f);
                ^
/tmp/guix-build-openboardview-7.3.drv-0/openboardview-7.3-checkout/src/openboardview/BoardView.cpp:197:16: error: ‘ImGuiCol_CloseButtonActive’ was not declared in this scope
   style.Colors[ImGuiCol_CloseButtonActive]    = ImVec4(0.98f, 0.39f, 0.36f, 1.00f);
                ^
/tmp/guix-build-openboardview-7.3.drv-0/openboardview-7.3-checkout/src/openboardview/BoardView.cpp: In member function ‘int BoardView::ConfigParse()’:
/tmp/guix-build-openboardview-7.3.drv-0/openboardview-7.3-checkout/src/openboardview/BoardView.cpp:320:9: error: ‘struct ImGuiStyle’ has no member named ‘AntiAliasedShapes’
   style.AntiAliasedShapes = false;
         ^
/tmp/guix-build-openboardview-7.3.drv-0/openboardview-7.3-checkout/src/openboardview/BoardView.cpp: In member function ‘void BoardView::ColorPreferencesItem(const char*, int, const char*, const char*, int, uint32_t*)’:
/tmp/guix-build-openboardview-7.3.drv-0/openboardview-7.3-checkout/src/openboardview/BoardView.cpp:527:32: error: invalid user-defined conversion from ‘ImColor’ to ‘const char*’ [-fpermissive]
  ImGui::ColorButton(ImColor(*c));
                                ^
In file included from /tmp/guix-build-openboardview-7.3.drv-0/openboardview-7.3-checkout/src/openboardview/Board.h:5:0,
                 from /tmp/guix-build-openboardview-7.3.drv-0/openboardview-7.3-checkout/src/openboardview/BoardView.h:3,
                 from /tmp/guix-build-openboardview-7.3.drv-0/openboardview-7.3-checkout/src/openboardview/BoardView.cpp:2:
/gnu/store/f4bs42wa4fgpid53xils7xk7mqglr0rg-imgui-1.66/include/imgui/imgui.h:1545:12: note: candidate is: ImColor::operator ImU32() const <near match>
     inline operator ImU32() const                                   { return ImGui::ColorConvertFloat4ToU32(Value); }
            ^
/gnu/store/f4bs42wa4fgpid53xils7xk7mqglr0rg-imgui-1.66/include/imgui/imgui.h:1545:12: note:   no known conversion from ‘ImU32 {aka unsigned int}’ to ‘const char*’
/tmp/guix-build-openboardview-7.3.drv-0/openboardview-7.3-checkout/src/openboardview/BoardView.cpp:527:32: error: too few arguments to function ‘bool ImGui::ColorButton(const char*, const ImVec4&, ImGuiColorEditFlags, ImVec2)’
  ImGui::ColorButton(ImColor(*c));
                                ^
In file included from /tmp/guix-build-openboardview-7.3.drv-0/openboardview-7.3-checkout/src/openboardview/Board.h:5:0,
                 from /tmp/guix-build-openboardview-7.3.drv-0/openboardview-7.3-checkout/src/openboardview/BoardView.h:3,
                 from /tmp/guix-build-openboardview-7.3.drv-0/openboardview-7.3-checkout/src/openboardview/BoardView.cpp:2:
/gnu/store/f4bs42wa4fgpid53xils7xk7mqglr0rg-imgui-1.66/include/imgui/imgui.h:416:29: note: declared here
     IMGUI_API bool          ColorButton(const char* desc_id, const ImVec4& col, ImGuiColorEditFlags flags = 0, ImVec2 size = ImVec2(0,0));  // display a colored square/button, hover for details, return true when pressed.
                             ^
/tmp/guix-build-openboardview-7.3.drv-0/openboardview-7.3-checkout/src/openboardview/BoardView.cpp: In member function ‘void BoardView::Preferences()’:
/tmp/guix-build-openboardview-7.3.drv-0/openboardview-7.3-checkout/src/openboardview/BoardView.cpp:746:59: error: call of overloaded ‘PushStyleColor(ImGuiCol_, ImColor)’ is ambiguous
   ImGui::PushStyleColor(ImGuiCol_Text, ImColor(0xff4040ff));
                                                           ^
In file included from /tmp/guix-build-openboardview-7.3.drv-0/openboardview-7.3-checkout/src/openboardview/Board.h:5:0,
                 from /tmp/guix-build-openboardview-7.3.drv-0/openboardview-7.3-checkout/src/openboardview/BoardView.h:3,
                 from /tmp/guix-build-openboardview-7.3.drv-0/openboardview-7.3-checkout/src/openboardview/BoardView.cpp:2:
/gnu/store/f4bs42wa4fgpid53xils7xk7mqglr0rg-imgui-1.66/include/imgui/imgui.h:256:29: note: candidate: void ImGui::PushStyleColor(ImGuiCol, ImU32)
     IMGUI_API void          PushStyleColor(ImGuiCol idx, ImU32 col);
                             ^
/gnu/store/f4bs42wa4fgpid53xils7xk7mqglr0rg-imgui-1.66/include/imgui/imgui.h:257:29: note: candidate: void ImGui::PushStyleColor(ImGuiCol, const ImVec4&)
     IMGUI_API void          PushStyleColor(ImGuiCol idx, const ImVec4& col);
                             ^
/tmp/guix-build-openboardview-7.3.drv-0/openboardview-7.3-checkout/src/openboardview/BoardView.cpp: In member function ‘void BoardView::ShowInfoPane()’:
/tmp/guix-build-openboardview-7.3.drv-0/openboardview-7.3-checkout/src/openboardview/BoardView.cpp:1124:19: error: ‘ImGuiWindowFlags_ShowBorders’ was not declared in this scope
                   ImGuiWindowFlags_ShowBorders | ImGuiWindowFlags_NoSavedSettings);
                   ^
/tmp/guix-build-openboardview-7.3.drv-0/openboardview-7.3-checkout/src/openboardview/BoardView.cpp:1182:60: warning: format ‘%d’ expects argument of type ‘int’, but argument 2 has type ‘std::vector<std::shared_ptr<Component>, std::allocator<std::shared_ptr<Component> > >::size_type {aka long unsigned int}’ [-Wformat=]
[... 161 lines cut ...]
                                                                                             ^
In file included from /tmp/guix-build-openboardview-7.3.drv-0/openboardview-7.3-checkout/src/openboardview/Board.h:5:0,
                 from /tmp/guix-build-openboardview-7.3.drv-0/openboardview-7.3-checkout/src/openboardview/BoardView.h:3,
                 from /tmp/guix-build-openboardview-7.3.drv-0/openboardview-7.3-checkout/src/openboardview/BoardView.cpp:2:
/gnu/store/f4bs42wa4fgpid53xils7xk7mqglr0rg-imgui-1.66/include/imgui/imgui.h:256:29: note: candidate: void ImGui::PushStyleColor(ImGuiCol, ImU32)
     IMGUI_API void          PushStyleColor(ImGuiCol idx, ImU32 col);
                             ^
/gnu/store/f4bs42wa4fgpid53xils7xk7mqglr0rg-imgui-1.66/include/imgui/imgui.h:257:29: note: candidate: void ImGui::PushStyleColor(ImGuiCol, const ImVec4&)
     IMGUI_API void          PushStyleColor(ImGuiCol idx, const ImVec4& col);
                             ^
[... cut ...]
/gnu/store/f4bs42wa4fgpid53xils7xk7mqglr0rg-imgui-1.66/include/imgui/imgui.h:1709:21: note:   candidate expects 3 arguments, 4 provided
/tmp/guix-build-openboardview-7.3.drv-0/openboardview-7.3-checkout/src/openboardview/BoardView.cpp: At global scope:
/tmp/guix-build-openboardview-7.3.drv-0/openboardview-7.3-checkout/src/openboardview/BoardView.cpp:4219:87: warning: unused parameter ‘io’ [-Wunused-parameter]
 inline bool BoardView::IsVisibleScreen(float x, float y, float radius, const ImGuiIO &io) {
````

piernov, 2018-12-06:
Yes, ImGui does break the API from time to time so OpenBoardView cannot be expected to build with every new ImGui release.
You can check my https://github.com/piernov/OpenBoardView/tree/wip/renderer-choice-2 branch for an OBV code that builds with ImGui 1.65.

Apteryks, 2018-12-07:
OK, thank you, I will experiment with your branch later :-)

piernov, 2019-07-26:
Closing this, one day an update will come with the newer ImGui, but there's no need to be able to build it with a newer version anyway.

Some fixes are still available here though: https://github.com/piernov/OpenBoardView/tree/wip/renderer-choice-2.1
Note that it works with v1.70 but it's broken again with v1.71…

closed 2019-07-26 by piernov