OpenBoardView/OpenBoardView #150: New feature request. Auto zoom to net.

ElectronicSupersonic, 2019-03-20:
Thanks for the software! I have a new feature suggestion.

Would it be possible to add an "Auto zoom to net" feature? The function (when enabled) would automatically zoom-in/zoom-out the view in a such manner so the whole selected net (upon clicking on to a net) becomes visible (taking the entire screen). I.E. instead of selecting a net, when zooming/panning manually to see the whole net, the feature would do it automatically.

Thanks!

pvtinflex, 2019-04-26:
Isn't this already part of OBV?  

Note the 'Zoom on selected net' in top-right of  Info panel.

[image]

ElectronicSupersonic, 2019-04-27:
Thanks for your reply!

The function you mention works in a different way. One must select a pin, when click its name in the info pan to show the whole thing.

The function I suggest would work in a different way. Say one wants to select a pin of an IC and see where the pin is connected to (or see all the spots/points where the signal is present), then upon clicking/double clicking the pin, view is automatically gets adjusted (zoomed in/out, panned) in a such manner so the whole net is visible (but still zoomed in as much as possible versus how the feature you mention does). Please see attached images, hopefully those will help to convey the idea.
[image]
[image]

Thanks!

pvtinflex, 2019-04-27:
Currently, if you click on the net/pin name in the right column, it should do the same thing.  

I suppose the difference is that the zoom level is not quite as "maximum" as perhaps you're wanting.   What I found in most use cases was that zooming to most of the screen realestate for the expanse of the net ends up being rather disorientating for net spans that don't go very far.  It's something I did experiment with in the past and found in daily use half the time I wasn't wanting it to do that, half the time I did.

Maybe it can be added as an optional behaviour, and it'll want to have the zoom level adjustable to suit the personal choice.

ElectronicSupersonic, 2019-04-27:
"Currently, if you click on the net/pin name in the right column, it should do the same thing."

My idea was to use clicking on the pin itself, rather clicking the pin, then clicking the net/pin name in the right column. Though it's more a matter of personal preferences/habits/background I guess.

As to the zoom... You are 100% correct about it being very much dependent on the net span. Perhaps there could/should be a user defined/set threshold of the net span at which the function would change its behaviour from zooming to not zooming (depending in the net size/span) and vice versa.

Thanks!

pvtinflex, 2019-04-27:
Would have to be a compound action to make it do that; perhaps akin to ctrl-alt-click or similar; at this stage not willing to yet bake in fixed actions.  Might have to wait for programmable user keys.

What I do need to do in the interim is provide a variable zoom factor entry in the preferences.

ElectronicSupersonic, 2019-04-27:
"...Might have to wait for programmable user keys."
It's obviously a feature "to be implemented in the next major version", since most likely would require substantial amount of work. So no rush here.

"...provide a variable zoom factor entry in the preferences."
That would be great.

Thanks!