OpenBoardView/OpenBoardView #156: Running OpenBoardView in TahrPup32v6.06

MrDuckGuy, 2019-05-01:
Hello,

I downloaded the OpenBoardView-R7.3-Linux.deb
file, and the system's 'PuppyPackageManager'
recognized it as new software and installed it.

The package installed without a menu entry.

When attempting to run this from a terminal 
window, my output was:

</tmp/runinterminal-22976: line 3: /usr/bin/openboardview: cannot execute binary file: Exec format error
Script completed hit RETURN to close window.>

I uninstalled the deb package and then tried installing
the rpm package. Again the PuppyPackageManager
recognized the new software and installed it.

The terminal output was: 

</tmp/runinterminal-29439: line 3: /usr/bin/openboardview: cannot execute binary file: Exec format error
Script completed hit RETURN to close window.>

... basically the same. I am obviously missing something.

Please if there is anyone who knows what I need to do
in order to run this program, let me know.

Thanks in advance, Kelikaku B'H.

▶—— Linux Kernel ——◀
Kernel Release: 4.1.30
Build Date: Fri Aug 12 18:33:11 BST 2016
OS Support: GNU/Linux
Architecture: i686
SMP Enabled: Yes
PAE Enabled: Yes

Distro: tahrpup 6.0.6
Window Manager: JWM v2.3.6
Desktop Start: xwin

pvtinflex, 2019-05-01:
I'd just try the build method;

apt-get install git build-essential cmake libsdl2-dev libgtk-3-dev

git clone --recursive 'https://github.com/OpenBoardView/OpenBoardView'

./build.sh

./bin/openboardview

piernov, 2019-05-01:
"Architecture: i686" may be the issue. Package is built for x86_64 I think. It i s not supposed to be ran on ancient systems as it require proper 3D acceleration. It may work, but will be unbearably slow without hardware acceleration.

pvtinflex, 2019-05-01:
Good spotting @piernov - didn't notice that at all.

MrDuckGuy, 2019-05-01:
Yes. I discovered that too. It's a 64-bit application, and
I was running a 32-bit OS.

I loaded it into the 64-bit version of the OS, and now
I seem to be missing a shared library: libSDL2-2.0.so.0?

--------------------------------
**/usr/bin/openboardview: 
error while loading shared libraries: libSDL2-2.0.so.0: 
cannot open shared object file: No such file or directory
Script completed hit RETURN to close window.**
-------------------------------

piernov, 2019-05-01:
The Debian package depends on `libsdl2-2.0-0`, which should provide `libSDL2-2.0.so.0`. Were the dependencies installed successfully?

MrDuckGuy, 2019-05-03:
I have a frugal install so I lack that library,
 and the installation package didn't include it. 

I have yet to try this link, to which I was referred
on another forum:

[http://archive.ubuntu.com/ubuntu/pool/universe/libs/libsdl2/libsdl2-2.0-0_2.0.2+dfsg1-3ubuntu1_amd64.deb](url)

I will update this thread if there is any result 
(other than my computer melting into a 
puddle of  useless goop) favorable or 
otherwise. B'H.

piernov, 2019-05-03:
I'd think your distribution package manager can fetch that dependency for you, and tell you that it is required when installing the OBV package.
At least under Ubuntu, `apt install ./OpenBoardView-R7.3-Linux.deb` takes care of that.

MattSturgeon, 2019-05-12:
IIRC using `dpkg -i` directly doesn't install dependencies - although I'd expect it to warn if one was missing... Try re-installing using `apt` as peirnov suggested.

You shound't have to manually install dependencies on any modern linux distro.

smoe, 2019-06-17:
Hello, I have created a Debian package for OpenBoardView and put that work on https://salsa.debian.org/electronics-team/openboardview . There is now glad in Debian unstable and the utf8.h library - I tend to think that technically the package is ready enough to be uploaded to the distribution. Please trigger me whenever you want that upload to happen.

piernov, 2019-07-26:
@smoe first, thanks for the effort. Could you create a new issue so that we can track it better?
I have a few concerns described below (I'll copy them on the new issue once it's created)

I don't use Debian so unfortunately I don't know the common practices there. Wouldn't it be better to preserve the upstream git history, also making it easier to track changes (as you can merge/rebase on the next release tag)? And rely on a "stable" release? I know there haven't been a release in a long time and there are a few fixes on master though.

About the description in `debian/control`, I'd like to clarify that OpenBoardView cannot display "schematics" themselves, it is more about rendering the PCB layout (but only showing parts and their pins, and the associated nets). I'd also like to remove the `and has quite some popularity on YouTube` sentence as it is not really relevant for a Debian package. About the feature list, it could be useful to list the supported file formats for example (although we don't even do it in the README…).

Last, could you clarify the purpose of `addCMakeLists.patch`?

closed 2025-08-20 by piernov