OpenBoardView/OpenBoardView #157: Open board file from command line

terrancewong, 2019-05-05:
# Just a feature request 

Open board file in command line args.

pvtinflex, 2019-05-05:
bin/openboardview  [-h] [-V] [-l] [-c <config file>] [-i <intput file>] [-x <width>] [-y <height>] [-z <fontsize>] [-p <dpi>] [-r <renderer>] [-d]
        -h : This help
        -V : Version information
        -l : slow CPU mode, disables AA and other items to try provide more FPS
        -c <config file> : alternative configuration file (default is ~/.config/OpenBoardView/obv.conf)
        -i <input file> : board file to load
        -x <width> : Set window width
        -y <height> : Set window height
        -z <pixels> : Set font size
        -p <dpi> : Set the dpi
        -r <renderer> : Set the renderer [ OPENGL1 = 1; OPENGL3 = 2; OPENGLES2 = 3 ]
        -d : Debug mode
pld@WorkstationTT:~/development/pld/OpenBoardView$

terrancewong, 2019-05-05:
Thanks~

What I did 
```
% openboardview --help
INFO: 
-------------------------------------------------------------
GL Vendor     : VMware, Inc.
GL GLRenderer : llvmpipe (LLVM 8.0, 256 bits)
GL Version    : 3.1 Mesa 19.0.2
GLSL Version  : 1.40
-------------------------------------------------------------
```
Let's print the usage page when user messing around  :-P

inflex, 2019-05-05:
https://github.com/OpenBoardView/OpenBoardView/commit/0382764180f7f7ace01311af32f2d48161b9a6e4

closed 2019-05-05 by inflex