OpenBoardView/OpenBoardView #158: Having some issues compiling this project in Windows

rlamosa1, 2019-05-23:
Good Morning,
Im new here and with this project, I was needing to make some changes so it be able to read Mentor Graphics files but when i try to compile it always gives me a lot of errors, i guess it must be related with Win7 x64, i saw somewhere that this project can only be opened for edition in fedora and other linux variant right?
Thanks in advance.

inflex, 2019-05-23:
Hi there,

  Are you referring to the binary version of the PADS files ( as opposed to ASCII  version) ?

  At the moment, OBV is primarily developed on a linux system and we cross compile to Windows ( either locally or use the automated Travis CI to do it for us ).

piernov, 2019-07-26:
Building under Windows is indeed not supported, there are a few things broken as of now (see #108 ). If you need to build a Windows executable, you have to set up a cross-compilation environment from Linux.
You can take inspiration from https://github.com/OpenBoardView/OpenBoardView/blob/master/.travis.yml#L34 .

Contributions are welcome if someone wants to fix this properly.

PiezPiedPy, 2019-08-25:
I've got a branch on my fork @ https://github.com/PiezPiedPy/OpenBoardView/tree/SDL2 I've just updated from the upstream/master.
I used CMake for windows to generate the VS solution. I compiled with VS2019 and the exe seems to be working fine (so far).

MattSturgeon, 2019-09-25:
@PiezPiedPy from a quick glance that looks good, would you submit a PR so we can merge your fixes upstream? Added brownie points if you add a appveyor (or whatever) CI to test windows builds!

PiezPiedPy, 2019-09-27:
@MattSturgeon PR sent 😉  
I have limited experience with CI, so regarding testing builds with CI, I wouldn't know where to start 😒