OpenBoardView/OpenBoardView #161: open boardview BRD files not opening

JLBPROFESIONAL, 2019-06-18:
Dear Sir .

I would like you to explain me some problems I have with the software openboard view. Some of the downloaded  BRD files from Vinafix or some other similar websites can not be opened with this programme and,  doubting it may be a  a problem of the  downloaded file itself, I have tried to open it with another programme such as boardviewer resulting a sucessful one.

I would like to know what this problem is due to. The same BRD file in two different computers does the same, that is, it does not open the file and it shows there has been a mistake.

I intended to purchase the flexbv proffesional but now I am not sure about it since it may arise the same problem when opening this kind of files; some can be opened succesfully and some others can not be opened.

I enclose the mistake reported by the programme with these files:

   "(this application has requested  the runtime to terminate it in an unusual way please contact        the application,s support for more information)"

I hope to have been clear enough trying to explain the matter and to be able to solve it so as to purchase the flexbv proffesional because I consider it to be much better and more accurate than some others I have already tested.

I look forward to hearing from you.
Best regards,

J.L.Bouzas

piernov, 2019-06-18:
For FlexBV support, please redirect your questions to its developer directly.

About the issue in OpenBoardView, we've seen some corrupted or inconsistent files in the past. OpenBoardView does some checks while opening the file to see if it appears to be consistent. If it is not, it'll show a similar error to what you're describing.
Can you tell us which files are causing issues? (or upload them here)

Also note that there are multiple different file formats that use the `.brd` extension. OpenBoardView does not support all of them, only "Landrex Test Link" and "TOPTEST BoardViewR4" variants (for the `.brd` extension, we support some other formats with other extensions). Pull requests to add file format support are welcome. Other software may support a different set of file formats.

inflex, 2019-06-18:
Hello @JLBPROFESIONAL , following on from @piernov . . . yes, the Allegro BRD binary format is one that is sometimes seen on the various download.  Typically you can distinguish them by the fact that their thumbnail contains what seems to be many layers of information compared to the TestLink & TopTest boards.

There is a bounty out for the decoding of the Allegro binary format but as yet no one has been able to present a solution for OBV (or FBV)

JLBPROFESIONAL, 2019-06-19:
this is an example of a file that can not be opened with open
boardview and if with boardviewer for example.
Is it possible with the flexbv professional version to open it?

piernov, 2019-07-26:
@JLBPROFESIONAL I don't see any example file, could you upload it again on the Github issue directly?

andreacero, 2020-11-24:
Hi,
I'm new here and I have the same issue...I attached the error image and the .brd file that I can't open with my boardview. 
Can you help me?
With the program phoneboard I can open the file but it's inpossible to work with it, so I think that the problem isn't in the brd file.
Thanks
Andrea
[image]
[https://drive.google.com/file/d/1wtL6WZosoeiDbTqWl0GfriTO7S1JWUDX/view](url)

piernov, 2020-11-24:
The link points to a MacBook boardview file which is supported by OpenBoardView and opens fine.
The screenshot shows an error about an iPhone boardview in PADS format which is not supported by OpenBoardView. (hint: you may find some compatible files for iPhone on badcaps.net)

andreacero, 2020-11-25:
I know that is the an error about an iPhone boardview but I use the screenshot to make an example of the error that I have with the mac boardview file linked up.
I use a PC with win 7.
All other Apple brd I'm able to open but not that.
Thanks for the answer...

andreacero, 2020-11-25:
Hey,
this is the real capture of my screen with the problem and the file that I have insert before.
Help me please.
[image]

piernov, 2020-11-25:
Remove the special character in the filename.

closed 2022-03-20 by piernov