OpenBoardView/OpenBoardView #162: Can you add support for. CAE format file?

my6010, 2019-07-26:
[image]

[Asrock 760GM-HDV_1.01.zip](https://github.com/OpenBoardView/OpenBoardView/files/3433872/Asrock.760GM-HDV_1.01.zip)
This is the official PCB file of asrock!

pvtinflex, 2019-07-26:
Any reference to the spec?  

Haven't seen CAE before.  Looks like a binary format, or at least some type of encoded text ( similar to BRD ).

pvtinflex, 2019-07-26:
Looks like it’s Dassaults propritory format used primarly in there Computer Aided Engineering (hence the CAE) format.

piernov, 2019-07-26:
This is most likely encrypted/compressed somehow, as there are no clear-text strings and the entropy is pretty high. Except for the last 4 bytes `$_BV` which may be the magic number.

Associated viewer software appears to be `PCBRepair Pro`. Which both by name and user-interface looks pretty similar to Asus' `PCBRepair Tool`. Asus and ASRock companies are historically linked.
Please upload the `PCBRepair Pro` software so that we can try to reverse it.

So my guess is that it is a variant of FZ format. @pafmaf was the one to decrypt it and find the key, so maybe he could take a look at it?

my6010, 2019-07-26:
[PCBRepairPro_Setup_3rd.zip](https://github.com/OpenBoardView/OpenBoardView/files/3436138/PCBRepairPro_Setup_3rd.zip)
This is the software that opens CAE format!

martonmiklos, 2019-08-06:
> So my guess is that it is a variant of FZ format. @pafmaf was the one to decrypt it and find the key, so maybe he could take a look at it?

Looks to be true, in the settings you need to specify the FZ File Path. 
[image]

(This is a directory where you need to put the CAE file to actually be able to load.)

pafmaf, 2019-08-07:
> So my guess is that it is a variant of FZ format. @pafmaf was the one to decrypt it and find the key, so maybe he could take a look at it?

I'll have a look when I find the time (it shouldn't be too hard to figure out).

calluchenko, 2023-09-04:
> [PCBRepairPro_Setup_3rd.zip](https://github.com/OpenBoardView/OpenBoardView/files/3436138/PCBRepairPro_Setup_3rd.zip) This is the software that opens CAE format!

u have new version sir?

goccanh1, 2023-09-04:
Hello!

i tried opening it with the available tool and everything doesn't work? Can you show me how to open it? How to bypass software copyright? The number of days of acceptance is 26 days, I set the time to 01/01/2020 but everything still hangs

Tilka, 2025-01-07:
CAE is just FZ with a different key. You'll need to remove the parity check.
```
0x477fa6a2 0xfb9b5e2b 0x77bcac57 0x2d7cef8c
0x69825182 0xfa231194 0x96ee6d48 0x520a9b74
0x0619cb60 0x95918dfb 0x1c829771 0x03f6655c
0xbba3b302 0xf3cbcc66 0xb42e9ac7 0x417b37dd
0x34854b8c 0xf95a9547 0x7950401e 0xc3271f83
0x0e7c9a6e 0xcfa7f799 0x616d9d05 0x200ac08f
0x7cdb242f 0x30d3bc5e 0x2983cc29 0x9da249c9
0x7509f015 0x6632580e 0x83247f04 0x6525ed71
0x02fa242a 0x47b12928 0x7ed51b5d 0xf69cd51b
0x66f24c77 0x042856b9 0x00e37970 0x88b6624d
0x6826cd76 0xd2a4c9fe 0x2eff487a 0x09648fae
```

Sample file posted above:

[image]