OpenBoardView/OpenBoardView #163: Native Debian Package (mostly?) ready for an official distribution

smoe, 2019-07-26:
Following the suggestion by @piernov in a comments to issue https://github.com/OpenBoardView/OpenBoardView/issues/156, hier the separate issue to track progress on a Debian Policy compliant Debian package that could (but does not need to) be uploaded to Debian and its derivative distributions like Ubuntu and Mint.

Concerning the patch https://salsa.debian.org/electronics-team/openboardview/blob/master/debian/patches/addCMakeLists.patch this is a substitute for the previous distributed CMake instructions for the glad source tree. That was previously redistributed with the OpenBoardView source and has become a Debian package on its own. I have forgotten the exact cause for the substitution. Instead, I should have patched the files refering to it. I presume this was the easiest at the time and I have then forgotten about that technicality.

This package has its roots in the Rossmann videos. He complained at least once about Linux not being functional enough (no key to jump to first page on a PDF, most effort will now go into finding that clip again) for him and as a Debian Developer I would like to eventually prove him wrong. But I need OpenBoardView first :o)

piernov, 2019-07-26:
Thanks again. Here is the main link to the Debian repo as a reminder: https://salsa.debian.org/electronics-team/openboardview

Apart from the couple of concerns I had below, I'm good with that, I think it is a good addition.

I don't use Debian so unfortunately I don't know the common practices there. Wouldn't it be better to preserve the upstream git history, also making it easier to track changes (as you can merge/rebase on the next release tag)? And rely on a "stable" release? I know there haven't been a release in a long time and there are a few fixes on master though.

About the description in `debian/control`, I'd like to clarify that OpenBoardView cannot display "schematics" themselves, it is more about rendering the PCB layout (but only showing parts and their pins, and the associated nets). I'd also like to remove the `and has quite some popularity on YouTube` sentence as it is not really relevant for a Debian package. About the feature list, it could be useful to list the supported file formats for example (although we don't even do it in the README…).

Maybe @inflex can comment? I think he's using Ubuntu also.

smoe, 2019-07-26:
> Apart from the couple of concerns I had below, I'm good with that, I think it is a good addition.

That is very nice to hear. I am (sadly) not using the package myself, which makes me just a bit nervous. So, yes, it would be very nice to have this synced with @inflex .

> I don't use Debian so unfortunately I don't know the common practices there. Wouldn't it be better to preserve the upstream git history, also making it easier to track changes (as you can [...]

This is historically grown, from some pre-github time. Debian is currently revisiting this all. As of today, Debian packaging should not touch anything outside the debian folder. And if it touches something, then it does so via some explicit patch in the debian/patches folder. This should give upstream some confidence in that the original code goes into the package. This is also very easy to update, with tarballs from github releases and also for interim releases from master if that is so desired. Alternatively, we could have the debian folder also on a branch in your github repository if that is so desirable. The downside is that it is a bit trickier e.g. to remove the git submodules - or, thinking about it, maybe it isn't. Tell me if you want this to move here, so we can give this a try.
 
> About the description in `debian/control`, I'd like to clarify that OpenBoardView cannot display "schematics" themselves, it is more about rendering the PCB layout (but only showing parts and their [...]

Fixed.

> About the feature list, it could be useful to list the supported file formats for example (although we don't even do it in the README…).

I happily follow the README with whatever change surfaces. But I happily invite to Debian's gitlab on https://salsa.debian.org and the pkg-electronics team to patch/edit the packaging instructions. Also, if there is other software that is still missing in Debian et al., please point me to it.