OpenBoardView/OpenBoardView #168: OpenBoardView wont open

EUWest1, 2019-12-09:
Hi Folks!
Just downloaded OBV and tried to start it. But nothing happens. Frame tries to start but immediatly it closes and no error code displays. 

My System:
WIN 10
Intel Core i5 2450
Intel HD Graphics 3000
DirectX Version 12

Hope anyone can back me up!

EUWest1, 2019-12-10:
Update: If i try to start OpenBoardView trough Windows command i get following Error:

DEBUG: Could not create GL context: Der Vorgang wurde erfolgreich beendet [means like "Process were successfully closed"].
ERROR: Failed to initialize OpenGL context: Could not create GL context: Der Vorgang wurde erfolgreich beendet.

piernov, 2019-12-10:
Try to update graphics driver.

EUWest1, 2019-12-10:
Latest driver is installed allready.
I´m using the latest stable Release Version from http://openboardview.org/

piernov, 2019-12-28:
Please try another OpenGL app to confirm if the issue is specific to OBV as I can't reproduce the problem.
You can also try to run OBV from the command line with the `-r1`, `-r2` or `-r3` parameters to select between one of the rendering backend. (OpenGL1, OpenGL3 or OpenGLES2)