OpenBoardView/OpenBoardView #171: There was an error opening the file: <path>.brd

dneise, 2020-01-31:
Hey ... great project!

I just built and installed it this on Ubuntu 18.04 and tried opening this brd file:
https://github.com/adafruit/Adafruit-STEMMA-Soil-Sensor-PCB/blob/master/Adafruit%20STEMMA%20Soil%20Sensor.brd

I get the error message: "There was an error opening the file: <path>.brd"

I tried starting the binary in debug mode like `./bin/openboardview -d` but I do not get any additional error messages neither on the command line ... nor in the GUI.

Could you tell me what else I could try to figure out the problem? I think the .brd file was generaded using EagleCAD.

Have a great day and thanks a lot.

dneise, 2020-01-31:
Update .. I looked at #133 and just to make sure the white spaces in the file name are not an issue, I just renamed the file to `test.brd`. This did not help. Still the same error msg.

inflex, 2020-01-31:
Eagle PCB isn't yet supported, but looks like that may change soon.

While eagle uses the 'brd' extension, the format is vastly different to the other brd formats that OBV reads in (and likewise FBV).

FlexBV will read files exported from Eagle in Fabricator, GenCAD, Hyperlynx though.

Paul.

dneise, 2020-01-31:
Ah okay .. sorry I didn't get that Eagle is not supported.

closed 2020-01-31 by dneise