OpenBoardView/OpenBoardView #174: Samsung CAD rendering issues — lots of parts missing

piernov, 2020-05-08:
Haven't taken the time to look at this issue, but opening the attached boardview with OpenBoardView 8.0 gives this:
[image]
While opening with the `CAD.EXE` software gives this:
[image]

Maybe something I broke during a merge.
Tagging @skvatboy , if ever he wants to take a look at it.

[BA41-02176A-Boardview.CAD.zip](https://github.com/OpenBoardView/OpenBoardView/files/4596453/BA41-02176A-Boardview.CAD.zip)

skvatboy, 2020-05-08:
Hi Piernov 
LINE 28620-28622 has invalid character 
[image]
It is way it failed to load the file.
I don't know what encoding to use Or is the file is corrupted ?
How do you deal with it on others file formats

skvatboy, 2020-05-10:
Hi
I think the file corrupted, i fix the invalid character and it load.

[BA41-02176AFIXcad.zip](https://github.com/OpenBoardView/OpenBoardView/files/4605877/BA41-02176AFIXcad.zip)

anshadedavana, 2021-01-13:
Support for Samsung CAD file is implemented in OBV ?

Release note for OBV ver 8.0 clearly says Samsung CAD is now supported but i can't open any of the Samsung boardview files in either ver 8.0 or the latest 8.95 beta release. Am i missing something here ?.

skvatboy, 2021-01-13:
Hi anshadedavana

Maybe MattSturgeon broke during https://github.com/OpenBoardView/OpenBoardView/pull/165

Try
line 160 `valid = current_block != 0;`
To
`valid = true;`

piernov, 2021-01-13:
Fix #206 merged, thanks for fixing the file I sent too, sorry I didn't see your answer earlier. Seems like it's all good now.

closed 2021-01-13 by piernov