OpenBoardView/OpenBoardView #179: 8.0 r645 - Selection dissapear when flip board from menu

psycodeoday, 2020-05-26:
Reproduction:
-select pad of capacitor, which have connection to the second side of board.
-now go to menu View and use Flip Board.
-your selection will disappear. You can evidently think, that this capacitor doesn't have any connection to this side of the board, which is not true.

I was fixing few boards in this way and it was so painful.

BTW:
Today I've read the Controls from Help and I saw that space key can also do Flip board, and it works - my selection not disappears when I flip board in this way.

Please fix it, because most of the people doesn't read manuals and are using menu instead of "secret" keyboard shortcuts.

Thank You.

piernov, 2020-05-26:
Hi, thanks for the report.

The menu item reacts on mouse button *push down*, that closes the menu, then boardview area reacts on mouse button *release*, that deselects the currently selected item (or selects another one).

This will be fixed in a future release once the ImGui upgrades are merged. ImGui's menu items behavior was changed in ImGui v1.60 so that the item gets activated on mouse button *release* rather than *push down*. The Windows binary over there: https://github.com/OpenBoardView/OpenBoardView/pull/176#issuecomment-628345413 contains the mentioned fix.

In the 8.0 release some keyboard shortcuts are shown next to the menu items. In a future release, keyboard shortcuts will be configurable. You can also use the `^` button in the menu bar that doesn't have this issue. The middle click of the mouse can be used to flips the board as well.

However there is another behavioral issue: when using the menu item or the `^` button, the boardview also gets flipped using Y mouse coordinate as a symmetry axis which is a pretty annoying effect that should not happen.
Flipping using Y mouse coordinate as a symmetry axis was probably not the intended behavior even when using the keyboard shortcut to begin with so I think that should be disabled altogether.

piernov, 2020-07-09:
f9989543321e0fd990e798fb94c46be4d0376057: `flipMode` is now set to 0 by default to not use the mouse pointer as rotation axis.

closed 2020-07-09 by piernov