OpenBoardView/OpenBoardView #182: feature request for adding -g <geometry> flag to control initial window size and placement

dithotxgh, 2020-06-23:
Fiddling around with two monitors here, and the current obv app always opens on the top monitor, in the center of the screen.  Sure enough, SDL_CreateWindow() is called with  SDL_WINDOWPOS_CENTERED for both the X and Y positions.

For unix like systems, the fix is pretty straight forward, just add a '-g'  command line flag that takes an X11 style geometry in parse_parameters(), and use the positions provided.  Without the command line flag, the current behavior is kept as default.

Attached is a first cut, sloppily formatted patch to accomplish the same for X11 based systems. I would look into doing the same for the WIN variant, but have absolutely zero experience in things the Redmond way.

dithotxgh, 2020-06-23:
I spun up a patch in https://github.com/OpenBoardView/OpenBoardView/pull/181

piernov, 2020-07-09:
Hi, thanks for the report and the contribution.

However I'm not quite sure this should be handled by OBV itself, it sounds like to me that it's the job of the window manager. I don't have your use case though. By default OBV indeed calls `SDL_CreateWindow()` with `SDL_WINDOWPOS_CENTERED`. I see that it `SDL_CreateWindow()` can take `SDL_WINDOWPOS_UNDEFINED` instead, which I'd guess doesn't try to force the window position. Maybe you could try that instead, and see if the window manager places the window as you expect it?

For the patch in itself, I'd prefer avoiding the dependency on bare X11 libs, as they're not expected to be available on Windows, macOS or Android for example. It seems like it is used to parse the geometry string with `XParseGeometry()`. We already have `width` and `height` parameters so we could just add a vertical pos and horizontal pos parameters to keep it simple.