OpenBoardView/OpenBoardView #184: Wayland and vulkan support

romanstingler, 2020-06-25:
most distros use wayland and vulkan is easier to code and faster than opengl. 
maybe there will be some rewrite one time

mmuman, 2021-11-14:
Apart from SDL2 which seems to have some Wayland support, the only mention of X11 is the manual loading of GTK, which is only used for the file picker. Maybe some portable libs would help there, like [nativefiledialog](https://github.com/mlabbe/nativefiledialog/). It also have a zenity fallback.

piernov, 2021-11-15:
That's only loading the GTK 2 library if GTK 3 is not available. GTK 2 doesn't support Wayland anyway. SDL2 should support it, but I never tried, Wayland never quite worked right for me.

About Vulkan, there's no reason to try to add support for it. OpenGL 1.2 and OpenGL 3.0 backends (with optional GLES2 support) cover everything we need. We rely on ImGui's implementation anyway so we don't write any OpenGL code ourselves, but Vulkan wouldn't make it easier.