OpenBoardView/OpenBoardView #187: Cant open .brd files from Altium?

jellewie, 2020-08-04:
I tried to export .brd files with altium (File->Export->IDF board) and tried different versions (Version 3.0 & Version 2.0) different formats (.brd+.pro & .brd+.lib) and with and without the use of Unicode.
None of the above-mentioned formats seem to work. All of them give the error "There was an error opening the file: Location:\folder\file.brd"

Can this even work, if so what is the combination?

inflex, 2020-08-04:
Do you have the option to export to GenCAD, IPC-D-356, Hyperlynx, or fabmaster?

jellewie, 2020-08-04:
No it doesn't seem like it. As least not without add-ons

**Export**
- DXF/DWG
- IDF Board
- Protel Netlist
- Specctra Design
- Step 3D
- VRML
- PDF3D

**Fabrication Outputs**
- Report boards stack
- Composite drill guide
- Drill drawinfs
- Final
- Gerber files
- Gerber X2 files
- IPC-2581
- Mask set
- NC drill files
- ODB++ files
- Power-plane set
- Test point report

**Assembly:**
- Assembly drawings
- Generates pick and place files
- Test point report

inflex, 2020-08-04:
Could try the ODB++ .  That's quite frustrating that there's no GenCAD at least :(

Do add-ons cost, or are they just a matter of installing?

jellewie, 2020-08-04:
Found some more options:
Add ons seem to cost money as far as I can tell

**Gerber options:**
- Gerber
- ODB++ (Below)
- Netlist
- IPC-D-350 (doesn't work)
- Frill
- Save Frill
- Mill/Rout
- DXF
- Part Centroids
- Aperture list
- Aperture library
- Bitmap

ODB++ seems to make 1 big folder with subfolders, not sure what/how to load

jellewie, 2020-08-04:
Found more options in the save menu

**Save PCB**
- PCB bin files (.pcbdoc)
- PCB 3.0 bin files (.pcb)
- PCB 4.0 bin files (.pcb)
- PCB 5.0 bin files (.pcbdoc)
- PCB ascii file (.pcbdoc)

**Save Schematic**
Advanced schematic library (.schdoc)
Advanced schematic ascii (.schdoc)
schematic library 4.0 (.sch)
Advanced schematic template (.schdot)

**Save Gerber**
CAMtastic files (.cam)
Expoer CAMtastic! 2000 DE binary file (.cam)

inflex, 2020-08-04:
Try export as PCB ASCII (pcbdoc) and CAMtastic (first one, not the binary), any chance you can attach here?

jellewie, 2020-08-04:
Will for sure try tomorrow.
And I can't attach the real file, but I can try to make just a small example or something, and export that.

inflex, 2020-08-04:
Test board outputs will be fine, thanks.

piernov, 2020-08-04:
None of the format that can be exported from Altium are currently supported by OpenBoardView AFAIK. It might be possible to import the design files into KiCad and then export to a supported format using https://github.com/whitequark/kicad-boardview but something will most likely go wrong.

FlexBV supports HyperLynx (and in theory Protel but it didn't work for me) and that's an export format of Altium 17 (this issue page is about OpenBoardView though).

jellewie, 2020-08-05:
> Test board outputs will be fine, thanks.
[pcb ascii.PcbDoc](https://drive.google.com/file/d/13wPFTe2I-3KmrSPyqreZeUUGZ53Sxh21/view?)
[CAMtastic.Cam](https://drive.google.com/file/d/1IBpRROd9s8CANZmsx2_4E2BQH-UeJpsW/view?)
The board themself are a mess, but the export should have worked

inflex, 2020-08-05:
On FlexBV I can get a partially okay rendering, so I may look at doing a porting of the fileformat module to OBV (there'll be a lot of details dropped but at least it'll still provide a viable boardview ).

[image]

jellewie, 2020-08-05:
Well if there are any updates on this issue, keep me updated :D

inflex, 2020-08-05:
I forgot to ask, do you have a screen shot of the top/bottom of the original as it should be, so I can use it as a reference.  Many thanks!

jellewie, 2020-08-05:
Screetshot of both, top, bottom

https://drive.google.com/file/d/17Yk2Hk-hfCqvVJyw4xscrC-ChfCs077d/view

inflex, 2020-08-07:
First compile rendering, clearly some work needs doing on the outline type ( points vs segment variation ), but it's a start.
[image]

inflex, 2020-08-10:
Have done some parsing improvements, it's getting there, but I'll have to update a few things on OBV to pass on the ability to handle the outlines, as they're segment form in the export, as opposed to sequential points.  As strange as it sounds, it's actually quite difficult to convert from segments -> points and still produce the same outline with a concave polygon.   I have a few things which I might try in advance before modifying OBV's core engine but here's FBV's vew so far.

[
[image]
](url)

inflex, 2020-08-11:
Sorted out a few more parsing issues.  Very close now.  Once this is done, I'll be able to push back to OBV, while you won't retain outlines for the pads/parts it will still be functionally useful.

[image]

inflex, 2020-09-01:
@jellewie the latest OBV source should now be able to read your ASCII Protel / AD pcbdoc files.

jellewie, 2020-09-01:
> @jellewie the latest OBV source should now be able to read your ASCII Protel / AD pcbdoc files.

Yhea cool! will give it a go and try out some other schematics :D

mkunzecw, 2021-02-02:
Tried with the latest tag 8.95.0, but the viewer just closes if i try an 20MB pcbdoc saved as ASCII from altium 20.
```

$ ./openboardview-8.95.0-win32.exe  -d                                                                                                     
INFO: Initializing ImGuiRendererSDLGL3                                                                                                     
INFO:                                                                                                                                      
-------------------------------------------------------------                                                                             
 GL Vendor     : NVIDIA Corporation                                                                                                       
  GL GLRenderer : Quadro M600M/PCIe/SSE2                                                                                                 
    GL Version    : 3.2.0 NVIDIA 452.11                                                                                                     
   GLSL Version  : 1.50 NVIDIA via Cg compiler                                                                                             
   -------------------------------------------------------------                                                                         
     terminate called after throwing an instance of 'std::out_of_range'                                                                        
   what():  vector::_M_range_check: __n (which is 0) >= this->size() (which is 0)
--
```
Will try if i can get a file that crashes the same and i can share.

swiftgeek, 2022-08-23:
One more alternative path with importing to KiCAD: exporting as GenCAD should work now, which will provide rich description for parts.