OpenBoardView/OpenBoardView #188: CAMCAD file format

romanstingler, 2020-08-15:
Please add this unencryprted file format.

[test.zip](https://github.com/OpenBoardView/OpenBoardView/files/5079529/test.zip)

martonmiklos, 2020-09-02:
It's a standard gencad format see #110 . In my gencad branch you will find a sort of somewhat working gencad parser:
https://github.com/martonmiklos/OpenBoardView/tree/gencad

I have did a test with your samples and none of them got loaded. I will try to find some time to investigate it.

martonmiklos, 2020-09-02:
I managed to load test1.cad, however it has some area of improvement:

[image]

No board outline, parts rotated in an invalid way.

martonmiklos, 2020-09-03:
@romanstingler the test2.cad file seems to be truncated. The $BOARD section is not closed and there are several missing sections.

martonmiklos, 2021-01-02:
@romanstingler  FYI the code in my gencad branch reads your test1.cad file wihtout problems now.
(The bad part outline for the U4 is not related to the Gencad support).

The RSI-TRANSLATOR output has some non-standard conform things like in the COMPONENT - DEVICE relation it tends to use spaces in the device name whilst these should not be here and shall be replaced by underscore. Anyway my code corrects this while parsing.

larry15501, 2021-01-21:
More complex file (gencad exported by camcad)
[rv6.zip](https://github.com/OpenBoardView/OpenBoardView/files/5851496/rv6.zip)

martonmiklos, 2021-01-21:
Hi @larry15501 

Works well I suppose:
[image]

The loading takes some time, which could be improved by improving the parser. Now parses all GenCAD fields, however the OpenBoard needs only the fraction of them.

larry15501, 2021-01-22:
I didn't read error message before:
[image]

https://github.com/martonmiklos/OpenBoardView/blob/gencad/src/openboardview/FileFormats/GenCADFile.cpp#L28

larry15501, 2021-01-22:
It might be a good idea to display `SHAPE` from gencad file. It sometimes contains part number, when `DEVICE` field is missing it. It can also clarify footprint/size of the component.

martonmiklos, 2021-01-22:
> I didn't read error message before:
> [image]
> 
> https://github.com/martonmiklos/OpenBoardView/blob/gencad/src/openboardview/FileFormats/GenCADFile.cpp#L28

Daw. Call me a lazy pig :D  I left a hardcoded path in the source. 

> It might be a good idea to display SHAPE from gencad file. 

Yeah indeed. The problem is the fact that OpenBoardView does not have a concept of device shape, basically the rectangles you saw around the parts is "magically" generated from the pins position, count, refdes prefix. 

Anyway my plan was to add the gencad support as a fileformat support only to OBV as a first step. Later we could implement new features like arbitary part shape which might be beneficial for other file formats as well.

larry15501, 2021-01-22:
I meant just displaying string specified in `SHAPE`, just like the one from `DEVICE`.

galkinvv, 2021-01-24:
Tested gencad version with that hardcoded path replaced - it works! Great thanks!

Unlike other .fz files I opened earlier which lacks vias at all, the gencad opener seems to show vias as a test points. Vias are visible only on one side.

I've a gencad file and I don't know its correct rendering, but in OpenBoardView rendering it has a mechanical component drawn aside of main board. Maybe such rendering is really encoded in a file.
The open time was ~10 seconds, which is slow, but acceptable.
[image]

martonmiklos, 2021-01-24:
Hmm that screenshot looks weird.

If you can share that gencad I would like to check it in D2B.

What do you think about the vias? Having them only on one side is a bug of my implementation, however it might not make sense to display them at all.

galkinvv, 2021-01-24:
Shared that file via e-mail in your git commits.
About vias - for me they are pretty useful, displaying them on one side is MUCH better than not disaplying them at all.

galkinvv, 2021-02-21:
Also I found that all 2pin-elements on the back side are mirrored (X of last pin is swapped with X of last pin)

The problem is in the `parse_shape_pins_to_component` function.
I experimented with the `+=` VS `-=` sign in
```
pin.pos.x += tmpPos.x * cos_ - tmpPos.y * sin_;
pin.pos.y += tmpPos.x * sin_ + tmpPos.y * cos_;
```
see https://github.com/OpenBoardView/OpenBoardView/compare/master...martonmiklos:gencad#diff-c7c05d3df49b4e3d86c81b96b524e561421b1e2654502742836b577be4055d89R301

sometimes some parts (including the MEC part above) are rendered fine, but others become broken

So, there should be some non-trivial condition for selecting that sign for X and for Y (not just `part->mounting_side == BRDPartMountingSide::Top`)

WaseemAlkurdi, 2021-04-29:
@martonmiklos Any update on this?

martonmiklos, 2021-04-29:
Well I would like to find some time to implement some workarounds which listed here:
https://github.com/martonmiklos/OpenBoardView/issues
before start merging this to master.

Also the hacks mentioned above (hardcoded grammar) needs to be fixed too, and I will need to look into the issues mentioned by @galkinvv

galkinvv, 2021-04-29:
By the way I've used the cad-branch for several other cad files - and all were rendered fine.
So in general it is already very useful!
Оnly one of my files is problematic.

galkinvv, 2021-06-18:
I've fixed hardcoded grammar path - https://github.com/galkinvv/OpenBoardView/commits/gencad-embedbnf
(fill free to merge/cherry-pick this anywhere)

galkinvv, 2021-06-21:
And some more practicality improvements: 
-speeds up initial loading (8 seconds instead of 40 seconds for a huge file on a decent machine) [really, in my practical usage this bothered me more than incorrect orientation of some components]
-rebased on 8.95.1 (sorry, intermedite commits become incompilable)
https://github.com/galkinvv/OpenBoardView/commits/gencad-fasterload1

(as previousely, this all is based on great @martonmiklos work, fill free to merge/cherry-pick my additions anywhere)

martonmiklos, 2021-06-21:
Many thanks for your work @galkinvv !

I really like your speed improvements!
The current grammar covers all the elements in the standard (at least this is what I tried to acheive). 
I think a lot could be improved by skip the parsing of the non used elements (there are a lot since ht gencad includes all the cooper informations as well)

galkinvv, 2021-08-07:
> I've a gencad file and I don't know its correct rendering, but in OpenBoardView rendering it has a mechanical component drawn aside of main board. Maybe such rendering is really encoded in a file.

It seems that 
https://github.com/galkinvv/OpenBoardView/commits/gencad-rotation-fix-prototype fixes this.
The problem was in rotation angle calculation.

However, it maybe would be more correct use `MIRRORX`/`MIRRORY` of a `SHAPE` statement as a criteria instead of top VS bottom layer.

closed 2022-06-14 by piernov