OpenBoardView/OpenBoardView #193: OpenBoardView issue opening .brd files

MSI78-DF, 2020-10-11:
Good morning all,
I am new to OpenBordView, version 8.0.
I am already working with BoardViewer, version 1.2.0.0, which works fine.
With OpenBordView, however, I have a problem:
The software gives me an error every time I open a .brd file:
[image]

Does anybody have any idea what's going on?
Thanks in advance !
Sincerely,

piernov, 2020-10-11:
Hi, please try putting the file in a path without special characters (such as `é`).

MSI78-DF, 2020-10-11:
> Hi, please try putting the file in a path without special characters (such as `é`).

piernov, you're a god ;)
IIt works perfectly! Thank you so much!

Small mistakes make big headaches... 
I wish you a great day!

Thank you all. See you soon...

closed 2020-10-11 by MSI78-DF