OpenBoardView/OpenBoardView #194: Dell XPS 9560 .brd incompatible

MihaiZaim, 2020-10-19:
Hi! Would it be possible to view this boardview file using the OpenBoardView software? It gives me an error when I try to view it. The file has been created using the Cadence Allegro software. 

Thanks.

[Dell XPS 9560 LA-E331P R10.zip](https://github.com/OpenBoardView/OpenBoardView/files/5399140/Dell.XPS.9560.LA-E331P.R10.zip)

piernov, 2020-10-19:
Hi,
duplicate of #62 . The Allegro format is unfortunately not supported and will not be anytime soon. This is a very complex binary format without public specifications so it requires a lot of reverse engineering skills and work to figure out.
Use the Allegro Free Physical Viewer.

closed 2020-10-19 by piernov

MihaiZaim, 2020-10-19:
Hi! Understood, I'm used to working with OBV and I find the controls of Allegro extremely annoying and hard to use so I hoped it was compatible with OBV. Thanks for the reply, hope someone will make it compatible. 
Thanks.
    Pe luni,