OpenBoardView/OpenBoardView #196: 2 issues with the program

szero, 2020-11-29:
Build of the program I'm using: INFO: OFBV-BUILD: R661 OpenBoardView/master Sat Nov 28 13:40:36 2020
OS: Debian sid

1. Keyboard input is only debounced for OPENGL1 renderer but not for paste function
When you want to rotate the board for example, it spins randomly, when you want to paste component value into search box it pastes 2 or more times

2. When you got to File -> Program preferences and toggle slowCPU mode and toggle it back, the probe points turn back into circles but contact points on the components stay square.

piernov, 2020-11-29:
Thanks for the report. I pushed a fix for 1. (7f9ade38) and for 2. (4468c418).
For 1. I reproduced the issue only with the OpenGLES2 renderer. This is a bug that I actually identified a long time ago, but it seems like the commit got lost somewhere. The bug didn't affect OpenGL1 or OpenGL3 renderers.

szero, 2020-11-29:
Wow, best service for free software I've gotten so far, fix deployed in half an hour! Recompiled and its all fixed now, thanks!

closed 2020-11-29 by szero