OpenBoardView/OpenBoardView #198: OpenBoardView on Raspberry Pi?

SkipperDo, 2020-12-10:
Hi there... 

Is it possible to create a Version of the OpenBoardViewer for the Raspberry Pi? (ARM architecture)

Would be very cool...

piernov, 2020-12-10:
I will not be maintaining an ARM Debian package, but you can compile it yourself. On Ubuntu/Debian install the dependencies:
```
sudo apt install build-essential git cmake libsdl2-dev libgtk-3-dev libegl1-mesa-dev libgles2-mesa-dev
```
Clone and build the binary:
```
git clone https://github.com/OpenBoardView/OpenBoardView.git
cd OpenBoardView
./build.sh
```
Then `./bin/openboardview` to run it.
And you can build a Debian package if you want:
```
cd release_build
make package
```

SkipperDo, 2020-12-10:
Yeah Baby...

This works well...
But i can't build a Package... Looks like something is missing for build a package...

`pi@RasPi:~/OpenBoardView $ cd release_build/
pi@RasPi:~/OpenBoardView/release_build $ make package
[  7%] Built target imgui
[ 12%] Built target sqlite3
[ 15%] Built target glad-generate-files
[ 22%] Built target glad
[100%] Built target openboardview
Run CPack packaging tool...
CPack: Create package using DEB
CPack: Install projects
CPack: - Run preinstall target for: OpenBoardView
CPack: - Install project: OpenBoardView
CPack: Create package
CPack: - package: /home/pi/OpenBoardView/openboardview_8.0-1_armhf.deb generated.
CPack: Create package using RPM
CPack: Install projects
CPack: - Run preinstall target for: OpenBoardView
CPack: - Install project: OpenBoardView
CPack: Create package
CMake Error at /usr/share/cmake-3.13/Modules/Internal/CPack/CPackRPM.cmake:767 (message):
  RPM package requires rpmbuild executable
Call Stack (most recent call first):
  /usr/share/cmake-3.13/Modules/Internal/CPack/CPackRPM.cmake:1870 (cpack_rpm_generate_package)

CPack Error: Error while execution CPackRPM.cmake
CPack Error: Problem compressing the directory
CPack Error: Error when generating package: OpenBoardView
make: *** [Makefile:97: package] Fehler 1
`
I tried to install rpm build... (im stupid i think...)

`pi@RasPi:~/OpenBoardView/release_build $ sudo apt install rpmbuild
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.       
Statusinformationen werden eingelesen.... Fertig
E: Paket rpmbuild kann nicht gefunden werden.`

piernov, 2020-12-10:
```
 CPack: - package: /home/pi/OpenBoardView/openboardview_8.0-1_armhf.deb generated.
```
The DEB package was built successfully.

The RPM package couldn't be built since the RPM build tools aren't installed (part of the `rpm` package), but it does not matter since RPM is for RedHat/Fedora, not for Debian/Ubuntu. Basically you can ignore the error.

SkipperDo, 2020-12-10:
I just figured this out... ;) the *.deb file can be successfully installed on a Pi...

Thank You for your help... Great Job!!!

I have an idea for an update of the OpenBoardViewer.. Where should i post this? Here in the Issues section? or better in Pull Requests?

inflex, 2020-12-10:
> I have an idea for an update of the OpenBoardViewer.. Where should i post this? Here in the Issues section? or better in Pull Requests?

Probably ideal to create your own new issue for it if you want to ask/discuss it.

closed 2020-12-10 by SkipperDo