OpenBoardView/OpenBoardView #201: Can't make on debian

sturlan, 2020-12-17:
Hi,
I followed the instructions, it seems i cant install it on debian. Somebody can maybe point me in the right direction?
[image]

piernov, 2020-12-17:
Thanks for the report.
Building the pre-release (master branch) requires SDL2 2.0.12 which is available on Ubuntu 20.10 or Debian testing. Build compatibility with older SDL2 (thus older distributions) may be fixed in the future
For now a workaround could be to `git revert -X theirs ffca6ec3`.

piernov, 2020-12-17:
I built 8.95.0 on Ubuntu 18.04, hopefully that's good enough to run on Debian 10:
https://github.com/OpenBoardView/OpenBoardView/releases/download/8.95.0/openboardview_8.95.0-1_amd64.deb

sturlan, 2020-12-29:
OK. I had to delete the folder and git clone fresh, but it worked.

closed 2020-12-29 by sturlan

piernov, 2021-01-16:
Should be fixed by 077d12af.

Apteryks, 2023-04-01:
Hi!  I'm not sure why, but when building from source manually, I get this:
```
[env]$ cmake ..
OpenBoardView release: R921 OpenBoardView/master
-- Checking for std::filesystem
-- Checking for std::filesystem - available
-- Checking for module 'stb'
--   No package 'stb' found
-- Found FontConfig version 2.13.94
-- Found GTK version 3.24.30
-- Found GIO version 2.70.2
-- Configuring done
CMake Error at src/openboardview/CMakeLists.txt:157 (add_executable):
  Target "openboardview" links to target "SDL2::SDL2main" but the target was
  not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.
maxim@hurd ~/src/OpenBoardView/build [env]$ pkg-config --cflags sdl2
-D_REENTRANT -I/gnu/store/mjwhybz735zxmsm92acyzfv70i0md2ar-sdl2-2.26.2/include -I/gnu/store/mjwhybz735zxmsm92acyzfv70i0md2ar-sdl2-2.26.2/include/SDL2
```
As the above output shows, SDL2 is at 2.26.2.

Apteryks, 2023-04-01:
Nevermind my previous comment, it was picking up and old sdl package from the system apparently; building in a container fixes that.

commit 077d12af (2021-01-16): Imported namespace targets for SDL2 <2.0.12 support (closes #201, #205)