OpenBoardView/OpenBoardView #202: GLIBC_2.29' not found

sonichy, 2020-12-17:
9.0-alpha1,8.90.3 (alpha3) GLIBC_2.29' not found, GLIBCXX_3.4.26 not found
8.0-1 GLIBCXX_3.4.26 not found
R7.3: Open file Boards, what is the suffix ?

piernov, 2020-12-17:
> 9.0-alpha1,8.90.3 (alpha3) GLIBC_2.29' not found, GLIBCXX_3.4.26 not found
> 8.0-1 GLIBCXX_3.4.26 not found

These versions were built with Ubuntu 20.04 or 20.10. I will try to rebuild with Ubuntu 18.04. I don't think I will be able to support older distributions.

>  R7.3: Open file Boards, what is the suffix ?

I'm afraid I did not understand the question.

piernov, 2020-12-17:
8.0 and 8.95.0 have been rebuilt with Ubuntu 18.04 (glibc 2.27 & GCC 7.5):
https://github.com/OpenBoardView/OpenBoardView/releases/download/8.0/openboardview_8.0-2_amd64.deb
https://github.com/OpenBoardView/OpenBoardView/releases/download/8.95.0/openboardview_8.95.0-1_amd64.deb

sonichy, 2020-12-18:
openboardview_8.95.0-1_amd64.deb can run.
I know, open dialog, Boards -> .brd
Can you give some sample files to open ?
It can not open the file that boardviewer can open.
[image]

inflex, 2020-12-18:
Please attach/provide file you're trying to open

sonichy, 2020-12-19:
[MS7740.zip](https://github.com/OpenBoardView/OpenBoardView/files/5718815/MS7740.zip)

inflex, 2020-12-19:
I thought we had CAD support in OBV ?

I know it definitely reads in FBV ( see attached )

[image]

inflex, 2020-12-19:
Just realised, this is GenCAD, and no, we don't have support for that in OBV as yet. I might migrate the FBV one over

piernov, 2020-12-19:
OpenBoardView supports the "Samsung CAD" format, unfortunately not the GenCAD format used by MSI.

closed 2020-12-19 by sonichy