OpenBoardView/OpenBoardView #203: Change keyboard control "Search" from ctrl-f to cmd-f for mac version.

gerikerschi, 2020-12-25:
Please change the keyboard shortcut for "search" on the Mac version from ctrl-f to cmd-f.
Normally on Mac the search is always cmd-f, only in this software it is different.
Please change.

Thanks a lot!

JonasCz, 2021-01-14:
I'd love to also see mac UI integration in general: menu bar, pinch-to zoom and two finger pan...

piernov, 2021-01-15:
With OpenBoardView 8.95.0 you can change the key bindings to suit your preferences. I'll see if I can make Cmd+F the default for macOS rather than Crtl+F.

Gestures for zooming and panning will be added if it works the same as on Android.

Menu bar would be a lot of work with too little benefit especially since the interface is designed with its own integrated menu bar with buttons and checkboxes. Contributions are welcome nonetheless.

piernov, 2021-01-15:
I changed the default modifier from Ctrl to Cmd for macOS.

`SDL_MULTIGESTURE` does not seem to trigger on my MacBook Air 13" 2013 on Big Sur with SDL2 2.0.14 which is basically what would allow handling the gestures, so for now it's not gonna be implemented.

gerikerschi, 2021-01-16:
The feature has a great added value for me!
Thanks a lot!!!

closed 2022-03-20 by piernov

gerikerschi, 2023-02-16:
> pinch-to zoom

Please deactivate this function! I love to zoom with scroll function.

commit 4430e5c8 (2022-03-20): Enable gestures on multi-touch touchpads (closes #203)