OpenBoardView/OpenBoardView #205: Cant build

jaspi72, 2020-12-31:
jaspi@jaspi-aceraspire:~/OpenBoardView$ ./build.sh
Compiling OpenBoardView using 4 threads (release build)
Extra flags passed to CMake: -DCMAKE_INSTALL_PREFIX= 
-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
OpenBoardView release: R738 OpenBoardView/master
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Found ZLIB: /usr/lib/x86_64-linux-gnu/libz.so (found version "1.2.11") 
-- Could NOT find SQLite3 (missing: SQLite3_INCLUDE_DIR SQLite3_LIBRARY) 
-- Found Python: /usr/bin/python3.8 (found version "3.8.5") found components: Interpreter 
CHECK_STARTChecking for std::filesystem
CHECK_PASSavailable
-- Checking for module 'fontconfig'
--   Found fontconfig, version 2.13.1
-- Found FontConfig version 2.13.1
-- Checking for one of the modules 'gtk+-3.0;gtk+-2.0'
-- Found GTK version 3.24.20
-- Configuring done
CMake Error at src/CMakeLists.txt:106 (add_library):
  Target "imgui" links to target "SDL2::SDL2" but the target was not found.
  Perhaps a find_package() call is missing for an IMPORTED target, or an
  ALIAS target is missing?

CMake Error at src/openboardview/CMakeLists.txt:118 (add_executable):
  Target "openboardview" links to target "SDL2::SDL2" but the target was not
  found.  Perhaps a find_package() call is missing for an IMPORTED target, or
  an ALIAS target is missing?

CMake Error at src/openboardview/CMakeLists.txt:118 (add_executable):
  Target "openboardview" links to target "SDL2::SDL2" but the target was not
  found.  Perhaps a find_package() call is missing for an IMPORTED target, or
  an ALIAS target is missing?

CMake Error at src/openboardview/CMakeLists.txt:118 (add_executable):
  Target "openboardview" links to target "SDL2::SDL2main" but the target was
  not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

-- Generating done

Im using FerenOS 20.04

piernov, 2020-12-31:
See #201

piernov, 2021-01-16:
Should be fixed by 077d12af.

inflex, 2023-04-04:
Having this issue reemerge (?) with SDL2 2.24.0 on Ubuntu 22.10 not finding SDL2::SDL2main

SDL2 setup on the system;
```
$ sdl2-config --version --cflags --static-libs --libs
2.24.0
-I/usr/include/SDL2 -D_REENTRANT
/usr/bin/sdl2-config: 61: : Permission denied
/usr/lib//libSDL2.a -lm -lasound -lm -ldl -lpthread -lpulse-simple -pthread -lpulse -pthread -lX11 -lXext -lXcursor -lXi -lXfixes -lXrandr -lXss -ldrm -lgbm -lwayland-egl -lwayland-client -lwayland-cursor -lxkbcommon -ldecor-0 -lpthread -lrt
-lSDL2
```

Here's the fresh OBV pull build;
```
$ ./build.sh .
Compiling OpenBoardView using 12 threads (release build)
Extra flags passed to CMake: -DCMAKE_INSTALL_PREFIX= .
OpenBoardView release: R921 OpenBoardView/master
-- Checking for std::filesystem
-- Checking for std::filesystem - available
-- Checking for module 'stb'
--   Package 'stb', required by 'virtual:world', not found
-- Found FontConfig version 2.13.1
-- Found GTK version 3.24.34
-- Found GIO version 2.74.3
-- Configuring done
CMake Error at src/openboardview/CMakeLists.txt:217 (target_link_libraries):
  Target "openboardview" links to:

    SDL2::SDL2main

  but the target was not found.  Possible reasons include:

    * There is a typo in the target name.
    * A find_package call is missing for an IMPORTED target.
    * An ALIAS target is missing.

-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.
CMAKE FAILED
```

piernov, 2023-04-04:
Different issue, this is specific to SDL2 2.24.0 on some systems only. Probably not worth adding a workaround for that, works with newer versions of SDL2 (or older).