OpenBoardView/OpenBoardView #209: Installed OBV in Linux Mint 18 dosn't launch.

mcrven, 2021-03-28:
I installed OBV via "openboardview_8.0-1_i386.deb". No problem with install process, no warnings about dependencies.

I can sow the OBV icon but... when click on it... nothing hapen. No errors reported, no warnings...

Have any help?

Thamks

piernov, 2021-03-28:
1. Make sure you are indeed running a 32-bit distribution rather than a 64-bit one. Otherwise use the `openboardview_8.0-2_amd64.deb` file.
2. If you are indeed running a 32-bit distribution, please run `openboardview` in a terminal and paste the output.

mcrven, 2021-03-29:
OK, piernov;
Here the out for terminal run of openboaedview 8.0-1

mcrven@MCR-KM400-8235 ~ $ openboardview
openboardview: /lib/i386-linux-gnu/libm.so.6: version `GLIBC_2.27' not found (required by openboardview)
openboardview: /lib/i386-linux-gnu/libc.so.6: version `GLIBC_2.28' not found (required by openboardview)
mcrven@MCR-KM400-8235 ~ $ 

Now, in  /lib/i386-linux-gnu/ the mentioned libraries exists as symlinks to libm-2.23.so

I hope some solution,,,

piernov, 2021-03-29:
OpenBoardView does not support old versions of glibc.
The 32-bit build supports Debian 10. (glibc 2.28)
The 64-bit build supports Debian 10/Ubuntu 18.04 (or derivatives) and newer. (glibc 2.27)
I can try to rebuild it on an old distribution if I find the time but no guarantee.
Please consider switching to a recent distribution with 64-bit support.

mcrven, 2021-03-29:
OK guy...
I have no 64 bits boxes running by the time.

Yesterday I cloned a developer version from github... and it works... as for say something,
But only some keyboard commands reacts and really slow. Mouse operations doesn't not reacts properly, only central button, wheel, drag and X (exit).
Here the terminal out after start:

mcrven@MCR-KM400-8235 ~/OpenBoardView $ ./openboardview.sh
INFO: Initializing ImGuiRendererSDLGL3
INFO: 
-------------------------------------------------------------
GL Vendor     : VMware, Inc.
GL GLRenderer : llvmpipe (LLVM 6.0, 128 bits)
GL Version    : 3.3 (Core Profile) Mesa 18.0.5
GLSL Version  : 3.30
-------------------------------------------------------------
DEBUG: That operation is not supported
WARN: ImGuiRendererSDLGL3: Unable to enable VSync: That operation is not supported
mcrven@MCR-KM400-8235 ~/OpenBoardView $

Any solution?

piernov, 2021-03-29:
It is running with the software OpenGL renderer so probably your graphics drivers aren't working (or the GPU is so old that it doesn't even support OpenGL 1.2 properly).
On top of that, if the CPU doesn't even support 64-bit, it must be a very old CPU so software rendering will be insanely slow on it.
If you can't get 3D acceleration working, sorry, no solution other than getting slightly better hardware.
OpenBoardView should run properly on mainstream 15 years old hardware (Core 2 with Intel integrated graphics). Anything older than that is really pushing it.

mcrven, 2021-03-29:
OK, piernov...

This box has a MSI KM4AM-V MB, AMD Sempron 2400+ @ 1.67 GHz, 1 GB RAM, 80 GB ide HD and dual S.O.: WIN 7 and LinuxMint 18.

Today I installed OpenBoardView_8.95.0_win32, recently downloaded from your site and... "It Works".
No previous versions and variants was working on Windows. Maybe it is a data of some interest for any one.

I hope you will have the time to "I can try to rebuild it on an old distribution if I find the time but no guarantee."

By the moment I can work using the windows version 8.95.

Thank you for your patience and time...
See you...

closed 2025-08-20 by piernov