OpenBoardView/OpenBoardView #210: How to extract component footprint to use it in Altium?

exander77, 2021-04-15:
I am kind of lost how to extract components or at least their footprint to use them to design PCB, such as Altium.

piernov, 2021-04-15:
It is not possible, OpenBoardView is not designed for this kind of usage, it is only designed to display the boardview files.
Additionally, most boardview files don't embed footprint information, what you see on the screen is just a rendered outline according to the position of the pins and the type of part. Even if the file format embeds footprint information, OpenBoardView cannot use them at this point.

exander77, 2021-04-15:
@piernov I opened a few boards and at connectors, pins and their labels were there. If I were to experiment with the code to see what I can get, would you point to in the right direction codewise?

swiftgeek, 2022-08-23:
While there isn't enough data with many formats, I guess SVG/DXF export option should be possible, to provide some aid. But wouldn't work well for components, more for creating custom daughterboards/modboards.

As for connectors you can identify most by measuring pitch and abusing google images, and you will very extremely rarely see completely custom connectors.

exander77, 2022-08-23:
@swiftgeek Anything is better than nothing. Exporting the whole MB outline would be nice as well. I wanted to extract the docking connector for example to make a mini pcb over its pins.

swiftgeek, 2022-08-23:
Docking connectors are also possible to identify, with some significant effort (most often they come from reference boards anyway)

Michael-Lloyd, 2022-10-03:
I think people shouldn't be too quick to state that it is _not possible_, the standard BRD file format is parsable, this repo contains a program that does exactly that -- but not for the purpose that you intend. 

I have successfully parsed the BRD file grammar for use with KiCAD _schematics_, but KiCAD 6 uses s-expressions that are easy to generate (I used Haskell). 

To make the files useful, depending on the specific format, you need to implement a parser that produces at a minimum:

- Symbols
- Placement of symbols in schematic files (And, for this to be readable in any way, you can't dump them all at coordinate (0,0), you need to use a graphical network arrangement algorithm)
- Footprints (Non-trivial for production use-cases, but for _review_ purposes, the BRD file tells you exactly what the pins are and where they are placed, this can be trivial if you can generate an output in the correct grammar) 
- Placement of footprints in a PCB layout design file
- Linking of reference designators and pin nets between both the schematic and the layout file 

Thankfully, in KiCAD, so long as you can generate the PCB layout file correctly, it can produce a terse and unfriendly schematic, which you can then clean up yourself. 

From memory, there are KiCAD to Altium exporters and importers (as well as the reverse). It is achievable. 

To echo the original reply though, you should be aware that this is not something you can just go and use to get a PCB printed. You will need to clean a lot of things up.