OpenBoardView/OpenBoardView #212: What Encoding does the .brd file use?

Joebeazelman, 2021-05-09:
I'm studying the source code to see how I may contribute to this project. Overall, it is very clear and well structured. There's one part of the code, however, that has me completely puzzled.

The Allegro .brd file's format, in particular, is baffling to say the least. I tried to open it using a several editors on a Mac as well as some online file readers to no avail. The text shows up as garbled text, indicating the file is either obfuscated, encrypted, or encoded using an uncommon or proprietary scheme.

In the BrdFile.h and BrdFile.cpp files, there's method to "Fix the data UTF-8",suggesting the file is UTF-8. But it's clearly unreadable by all the editors I've tried. It appears Brd files store their code points as a single byte, instead of the more common w_char (2 byte character). The code reads a byte into a two byte int, performs some bit operations, then casts it back into a single byte char. It's obviously not UTF-8, if the Unicode standard means anything.

What format is this file using? Why did they choose this format? If the purpose is for obfuscating their IP, they clearly failed at it. Could it be a scheme for compacting the file size. Any thoughts?

piernov, 2021-05-09:
The Cadence Allegro format is not supported. It is a pretty complex binary format for which the specifications are not available and that has not been reverse engineered successfully. 

The `.brd` file formats supported in OpenBoardView are entirely unrelated. They are pretty simple text formats used by the Landrex TestLink software (BRD) or the TopTest BoardView R4 software (BRD2).

The Landrex TestLink format has some sort of encoding for obfuscation, it is decoded by the few lines at https://github.com/OpenBoardView/OpenBoardView/blob/master/src/openboardview/FileFormats/BRDFile.cpp#L108 .
The `.bdv` format for the Honhan BoardView software is also a text format and uses some sort of obfuscation as well but even simpler, decoded by the `decode_bdv()` function at https://github.com/OpenBoardView/OpenBoardView/blob/master/src/openboardview/FileFormats/BDVFile.cpp#L10.
The `.fz` format for Asus' PCBRepairTool is still a text format, but it's compressed and encrypted so of course it was more complicated to deal with, but thankfully pafmaf managed to extract the encryption key.
The `.cst` format is a binary format but has no obfuscation and is nicely constructed, so it wasn't too difficult to reverse engineer (but still required some time).

UTF-8 is unrelated. Also UTF-8 is 1 to 4 bytes, with all ASCII characters being encoded on 1 byte matching with the usual ASCII table (0 to 127).

The `WCHAR` format is not "more common". It's really only used for the Unicode variant of the Windows API (and internally by Java but the developer never sees it). It is a 2 to 4 bytes UTF-16 format. No file format uses it. In OBV, it is only relevant when dealing with native OS API calls for example for paths. Note that in OBV this is not supported properly due to some limitations of using mostly portable code and MinGW for cross-compilation, so file paths with special characters in the path are not supported. This is an area that needs improvement.

Joebeazelman, 2021-05-13:
I isolated  and tested the section of code responsible for decoding the file and it only decodes certain characters. The expected result for the first line `#�c(�d&��q` should be  `str_length`, instead I get `s@	r_@	ng@	@	:
`

Is there another section of code I maybe overlooking?  Here's my test in Compiler explorer:

https://gcc.godbolt.org/z/3EYhE5fex

piernov, 2021-05-14:
You can't just copy/paste non-printable characters in your source code.

Also I'm sorry to say that we don't have the resources to provide individual support for general programming questions.

Joebeazelman, 2021-05-15:
Before you jump on your high horse and dismiss my inquiry as a "general programming question", perhaps you should at least be mindful that your C++ skills are stuck in the 90s as demonstrated by your heavy use of unsafe raw pointers and primitive memory allocation functions. Both are highly discouraged and offer no performance or size benefits. Perhaps you should ask me some "general programming questions" about modern C++.

As far as the code, I had already tried the code on a brd file with the same results. I wanted to include it, but I wasn't sure the online compiler supported. The characters in the string are not "unprintable" characters, they are merely ASCII representations of 1-byte ASCII values. How they are displayed and printed is left to the medium, hence the differences of how the strings are printed in the console and the editor. The compiler doesn't alter the strings data unless the encoding is specified with a prefix.

Here's what the standard says:

> Phase 5
> 1) All characters in character literals and string literals are converted from the source character set to the execution character set (which may be a multibyte character set such as UTF-8, as long as the 96 characters of the basic source character set listed in phase 1 have single-byte representations).
> 2) Escape sequences and universal character names in character literals and non-raw string literals are expanded and converted to the execution character set. If the character specified by a universal character name isn't a member of the execution character set, the result is implementation-defined, but is guaranteed not to be a null (wide) character.
> Note: the conversion performed at this stage can be controlled by command line options in some implementations: gcc and clang use -finput-charset to specify the encoding of the source character set, -fexec-charset and -fwide-exec-charset to specify the encodings of the execution character set in the string and character literals that don't have an encoding prefix (since C++11), while Visual Studio 2015 Update 2 and later uses /source-charset and /execution-charset to specify the source character set and execution character set respectively.

https://en.cppreference.com/w/cpp/language/translation_phases#Phase_5

inflex, 2021-05-15:
> Before you jump on your high horse and dismiss my inquiry as a "general programming question", you should at least be mindful that your C++ skills are stuck in the 90s as demonstrated by your heavy use of unsafe raw pointers and primitive memory allocation functions. Both are highly discouraged and offer no performance or size benefits. Perhaps you should ask me some "general programming questions" about modern C++.

You're probably talking about my code. I'm the guy from the 70's who mostly writes C and sprinkles in some C++ for convenience.

I appreciate your frustration but it'll be more conducive to avoid that sort of verbal trash/insult talk. Stick to the pragmatic, thank you.

Have you copied off an existing file decoding module to create one specifically for the Allegro BRD and stripped it back to template/outline form and started to write new code to deal with the unique nature of the Allegro BRD file?  Existing decoding functions in the other modules will not have any relevance with the attempts to decode the Allegro binary format ( other than formatting the decoded data in to the vectors for passing back to the boardview program.

Do you have some code files to share?

inflex, 2021-05-15:
Having a look at your submitted code;

First thing, as with what piernov mentioned, you can't just cut-paste that "string" in there, you need to set that up as a byte sequence another way, because you've lost all the information in the cut-paste of what it was originally;

ie;   uint8_t testchar[]= { 0x23, 0x52, 0x82, 0xF1, ..... }  // a more robust way of replicating what is in the Allegro data file we're reading

`if (!(x == '\r' || x == '\n' || !x)) `

What are you trying to detect with the !x portion ?

`            x =   (char) ~(((c >> 6) & 3) | (c << 2));`

What is the function you're trying to apply to the byte in this sequence?  Inversion of the byte rotated left two bits?

        
```
// Online C++ compiler to run C++ program online
#include <iostream>
#include <bitset>
#include <string> // for string class
#include <climits>

using namespace std;

string decode_board_str(string);

int main() {
    cout << "Encoded string: " << "#�c(�d&��q" << endl;

    cout <<  "Decoded:" << decode_board_str("#�c(�d&��q") << endl;
    cout << "Expected: str_length";
    return 0;
}

string decode_board_str(string str)
{
    for (int idx = 0; idx < str.length(); idx++)
    {
        char x = str[idx];
        if (!(x == '\r' || x == '\n' || !x)) 
        {
            int c = x;
            x =   (char) ~(((c >> 6) & 3) | (c << 2));
            //std::cout << x << std::bitset<CHAR_BIT>(x) << '/' << std::bitset<16>(c) << ' ';
        }
        str[idx] = x;
    }

    return str;
}
```

closed 2021-05-15 by piernov

Joebeazelman, 2021-05-19:
Thanks for your response. It turns out my compiler treated the char type as 16-bit due to a default option. I originally had the string as byte array of hex values, but figured cutting and pasting a few lines from the ".brd" file would be more demonstrative concerning my issue. Unfortunately, my editor encodes it correctly, the online editor doesn't, even though the strings appear identical on both. 

For the record, the previous response came across as very rude and arrogant. While he may not have the time to assist with what he perceives to be a newbie question, he wrote it on the behalf of all participants involved in the project including yourself. I think anyone would have reached the same conclusion albeit with a different reaction. Whether my response was constructive or not doesn't really matter when I'm told my interaction is unwelcomed.

AlexeyInwerp, 2021-05-19:
it might worth mentioning that FlexBV != Openboardview. 
now let me grab some popcorn and see how you try to lecture Paul L Daniels about his C coding style.