OpenBoardView/OpenBoardView #224: BVR Format isnt working

HeyWatchOutDude, 2021-11-15:
Hi,

is the ".BVR" format not supported by the "OBV" software?

Cant open it.

inflex, 2021-11-15:
What's the first line in the BVR file?  I'm guessing it'll be a BVR3 version file, as opposed to BVR1 that OBV currently has implemented. 

BVR3 needs to be ported over but there's other items in my queue that I have to sort out first.

HeyWatchOutDude, 2021-11-16:
Example file: 820-02020_V1.0.bvr

inflex, 2021-11-16:
Yes, that's a version-3 BVR  file, likewise will be the case with the 820-02016 board and any other new ones that come out.

BVR (v3) is in FlexBV currently but needs to be backported to OBV, or someone else can write their own decoder for OBV ( it's relatively trivial, BVR v3 is explicitly verbose in order to make parsing extremely simple )

HeyWatchOutDude, 2021-11-16:
Will it be supported in the future?

inflex, 2021-11-16:
I dare say it will be, yes.  Just a case of finding the time to do it ( I'm busy working on Teboview currently )

HeyWatchOutDude, 2021-11-16:
Ok thanks for your help so far :)

Tedworthy, 2022-04-18:
I was trying to access the exact same BV3 file yesterday for the A2338 MacBook Pro, but didn't want to purchase FlexBV just to open a single board. After reading the comments here, then comparing the BVR3 file with a BVR1 file to understand the format, I thought I would try to write a decoder for BVR3.

I'm pleased to say that I was successful and have been able to open this BVR3 file and see the board. All the parts and pins appear to be loaded perfectly, but I have come across an issue with the board outline ('format' in BRDFile). You can see the extent of the issue by checking the screenshot of OBV below and then comparing against a physical PCB photo from iFixit:

[image]
[image]

The board outline is defined at the end of the BVR3 file with the prefix "OUTLINE_SEGMENTED". I'm unsure of the format of this but it appears to be a set of points in the style "x1 y1 x2 y2 x3 y3...", so that is how the code is currently parsing them. This can't be far off because the board outline is largely correct, but something is clearly not right. I have noticed that under this format assumption, the outline path seems to return to previously visited points frequently, which is something I am yet to understand fully.

Here's a snippet of the problematic section of the file:
`OUTLINE_SEGMENTED 7966.000 3106.000 7307.000 3109.000 2875.000 3453.000 1344.000 3456.000 1253.000 3319.000 1253.000 3137.000 1344.000 3456.000 1253.000 3319.000 1219.000 3106.000 219.000 3106.000 1219.000 3106.000 1253.000 3137.000 128.000 3006.000 125.000 1962.000 219.000 3106.000 128.000 3006.000 97.000 1931.000 7.000 1931.000 0.000 1912.000 0.000 1800.000 97.000 1931.000 125.000 1962.000 7.000 1931.000 0.000 1912.000 22.000 1775.000 438.000 1775.000 0.000 1800.000 22.000 1775.000 532.000 1681.000 1057.000 94.000 532.000 1681.000 438.000 1775.000 1119.000 47.000 1482.000 47.000 1057.000 94.000 1119.000 47.000 3310.000 3.000 5560.000 0.000 2047.000 84.000 3244.000 84.000 2047.000 84.000 1482.000 47.000 3278.000 50.000 3282.000 22.000 3278.000 50.000 3244.000 84.000 3282.000 22.000 3310.000 3.000 5622.000 65.000 5619.000 440.000 5560.000 0.000 5622.000 65.000 5653.000 475.000 7644.000 475.000 5653.000 475.000 5619.000 440.000 7688.000 515.000 8003.000 515.000 7688.000 515.000 7644.000 475.000 8060.000 1384.000 8063.000 2247.000 8003.000 515.000 8063.000 562.000 8063.000 562.000 8060.000 1047.000 8032.000 1075.000 8060.000 1047.000 8032.000 1075.000 8028.000 1353.000 8028.000 1353.000 8060.000 1384.000 8032.000 2290.000 8028.000 3044.000 8032.000 2290.000 8063.000 2247.000 8028.000 3044.000 7966.000 3106.000 6957.000 3109.000 5294.000 3112.000 4607.000 3112.000 3869.000 3112.000 7278.000 3084.000 7272.000 3050.000 7307.000 3109.000 7278.000 3084.000 6985.000 3084.000 6957.000 3109.000 6985.000 3050.000 6985.000 3084.000 7247.000 3025.000 7013.000 3025.000 7247.000 3025.000 7272.000 3050.000 6985.000 3050.000 7013.000 3025.000 4632.000 3097.000 4607.000 3112.000 4632.000 3097.000 4638.000 3090.000 4632.000 3097.000 4638.000 3090.000 5294.000 3112.000 5266.000 3090.000 3528.000 3134.000 2875.000 3453.000 3528.000 3134.000 3544.000 3122.000 3816.000 3034.000 3572.000 3034.000 3869.000 3112.000 3838.000 3075.000 3816.000 3034.000 3838.000 3075.000 3547.000 3059.000 3572.000 3034.000 4632.000 3097.000 5266.000 3090.000 3547.000 3106.000 3547.000 3059.000 3544.000 3122.000 3547.000 3106.000`

I don't have a copy of FlexBV but if anyone does, I would appreciate it if they could open the above BVR3 file in that to see what the outline of the board looks like, and ideally posting a screenshot. I'm assuming the problem is with my understanding of this section of the file, but there is an outside chance it's actually an error in the original file and the board will look the same in FlexBV.

I'd like to get this issue resolved before raising a PR. If anyone is interested the code for this change is currently in a branch here: https://github.com/Tedworthy/OpenBoardView/tree/bvr3-file-format. Anyone wanting to view BVR3 files with OBV can check this out and compile their own version of OBV with this capability, as everything else is functional, it's only the outline of the board is likely to be incorrect. The instructions for doing so in the readme are really straightforward.

inflex, 2022-04-18:
The outlines are segment ( x1,y1 - x2, y2 ) and not gauranteed to be sequential in outline traversal.  You'll need to update the board outline function to have a segmented mode or you can write a sequencer that'll take the array of point pairs and sequence them by looking for the next pair that has the closest end point and merging them in to a sequential generic poly.

Tedworthy, 2022-04-18:
Thanks so much for that, the data certainly makes a lot more sense now. I will update the code to reflect that approach and raise a PR if all is successful.

HeyWatchOutDude, 2022-04-18:
@Tedworthy I have installed your application (build/run was successful) ... when I open a file with format ".BVR" is still getting the error message "unrecognized file format".

Tedworthy, 2022-04-18:
@HeyWatchOutDude OK, did you take the code from the branch bvr3-file-format? Regardless, let me know which board file you're trying to open and I'll give it a go myself. Would be a useful test case as I've only checked a single file so far.

Tedworthy, 2022-04-23:
For anyone who is interested, I've managed to get the outlines to work perfectly as far as I can see from testing. I have therefore raised a PR: #236

HeyWatchOutDude, 2022-04-24:
@Tedworthy I have used the following file "820-02020_V1.0.bvr" (A2338 | MacBook Pro M1)

Tedworthy, 2022-04-24:
@HeyWatchOutDude OK that's the same file I was using, and I can confirm it opens fine. I'm not sure where things are going wrong but it sounds most likely that the OBV version you're using does not contain the BVR3 code.

I suggest you go to https://github.com/Tedworthy/OpenBoardView/tree/bvr3-file-format and make sure you have downloaded or cloned that particular branch. You will know you have got the correct code if you can find the file src/openboardview/FileFormats/BVR3File.cpp. Then build using build.sh and run the resulting executable, and that version will successfully open that board file.

Tedworthy, 2022-04-24:
@inflex Could you possibly help me understand a few more things about the BVR3 format please? Most of these are off the back of comments in the PR above.

- In the first line of BVR3 files, e.g. 50.50.BVRAW_FORMAT_3, there are 2 numbers before the format indicator. What do these signify? Are they perhaps settings or something like multipliers for other values in the file?
- What are all the possible values for PART_SIDE? I have personally seen only T and B whose meanings are clear, but apparently 0 is also another value that's been seen. If that is valid, what does 0 mean?
- What is PART_ORIGIN used for? I gather it's a point but on most files it seems to be 0. 
- Similar to the above, what are all the possible values for PIN_SIDE? I have only seen 0 for this, but I'm unsure if that translates to T or B, or maybe it means inherit the same side as the part that the pin is in?
- What does PIN_TYPE mean and what are the possible values? Again I have only seen 2 as the value for this and I'm unsure what that signifies.

HeyWatchOutDude, 2022-04-24:
@Tedworthy could you please provide me your .bvr file, maybe mine is bad?

Tedworthy, 2022-04-24:
@HeyWatchOutDude I got mine from the Badcaps forum. I don't want to attach the file here just in case of copyright issues, but if you search 'badcaps 820-02020' in Google, the first result is the thread, and somebody attaches the board about halfway down the first page.

inflex, 2022-04-25:
> @inflex Could you possibly help me understand a few more things about the BVR3 format please? Most of these are off the back of comments in the PR above.

> 
> * In the first line of BVR3 files, e.g. 50.50.BVRAW_FORMAT_3, there are 2 numbers before the format indicator. What do these signify? Are they perhaps settings or something like multipliers for other values in the file?

  Not actually sure why they're there.  They're not part of the format; however they're rather "harmless" as it were as they do not interfere with the decoding.
  
> * What are all the possible values for PART_SIDE? I have personally seen only T and B whose meanings are clear, but apparently O is also another value that's been seen. If that is valid, what does O mean?

 O = both.  Aside from that, nothing else.
 
> * What is PART_ORIGIN used for? I gather it's a point but on most files it seems to be 0.

That's used if the pin placements are defined as part of a package, so the actual locations will need to be relative to the PART_ORIGIN.  In the current BVR files this isn't really used.  It's more of a "future thing".

> * Similar to the above, what are all the possible values for PIN_SIDE?

  As above.
  

> * What does PIN_TYPE mean and what are the possible values? Again I have only seen 2 as the value for this and I'm unsure what that signifies.

  Future planning.
  
  I'm presuming you're just filtering through a FlexBV binary to see the various tags?  There'll likely also be a few deprecated or defunct ones too.

Tedworthy, 2022-04-28:
@inflex Thanks a lot for that info, much appreciated. We were actually just going through various BVR3 board files from Badcaps testing the parsing code and checking the meanings of the tags, as part of the PR I raised. I'll update the code to handle the O/both case on part and pin side, and will make comments against the other tags in the parsing code for future reference.

HeyWatchOutDude, 2022-08-14:
@Tedworthy Hi, I have tried to build your application somehow the "build.sh" and "openboardview.sh" always ends with "MAKE INSTALL/STRIP FAILED".

barradao, 2022-12-17:
@Tedworthy Hello! I already downloaded your modification but I don't know how to build the app. Could you provide me with the exe already made? Or explain to me how can I do it? Thanks