OpenBoardView/OpenBoardView #228: [packaging] update imgui copy to a released version

Apteryks, 2022-01-30:
Hello,

The current bundled (as a submodule) copy of imgui is not on a release version (it's somewhere later after 1.79) and using 1.80 doesn't work.  There seems to have been recent changes in 1.86 (GL default loader selection code) that would perhaps make it easier to use as a system library.

So consider this ticket as a request to update the bundled copy of imgui to a released version, preferably to latest.

PS: openboardview will soon be available to install in Guix (which, like flatpak, can be used on any GNU/Linux distribution); c.f.: https://lists.gnu.org/archive/html/guix-patches/2022-01/msg01880.html.

Thank you :-)

piernov, 2022-02-06:
Yes, ImGui was set to a commit after their `tables` branch got merged into `master`, but at that time there weren't any new release after the merge yet. I may look into updating it if ever I make a new stable release of OBV, but in the meantime patches are welcome.

closed 2022-03-20 by piernov

Apteryks, 2022-03-21:
Thanks, I was able to use the system ImGui to build OpenBoardView for GNU Guix.

Apteryks, 2022-03-21:
See: https://github.com/OpenBoardView/OpenBoardView/pull/233/.

commit cbf99bb9 (2022-03-20): Update imgui to v1.87 (closes #228)