OpenBoardView/OpenBoardView #229: [reproducible-builds] Timestamp baked in executable / SOURCE_DATE_EPOCH not honored

Apteryks, 2022-01-30:
Hi,

While packaging this for Guix, I noticed the use of a __TIMESTAMP__ file that causes the OpenBoardView build to not be reproducible:

```
$ diffoscope /gnu/store/yc7shzgjhcf5jwf89fr9536zj563d0gb-openboardview-8.95.1{,-check}
--- /gnu/store/yc7shzgjhcf5jwf89fr9536zj563d0gb-openboardview-8.95.1
+++ /gnu/store/yc7shzgjhcf5jwf89fr9536zj563d0gb-openboardview-8.95.1-check
│   --- /gnu/store/yc7shzgjhcf5jwf89fr9536zj563d0gb-openboardview-8.95.1/bin
├── +++ /gnu/store/yc7shzgjhcf5jwf89fr9536zj563d0gb-openboardview-8.95.1-check/bin
│ │   --- /gnu/store/yc7shzgjhcf5jwf89fr9536zj563d0gb-openboardview-8.95.1/bin/.openboardview-real
│ ├── +++ /gnu/store/yc7shzgjhcf5jwf89fr9536zj563d0gb-openboardview-8.95.1-check/bin/.openboardview-real
│ │ ├── strings --all --bytes=8 {}
│ │ │ @@ -3369,15 +3369,15 @@
│ │ │  ##BoostPins
│ │ │  Boost (Pins)
│ │ │  ##BoostParts
│ │ │  Boost (Parts)
│ │ │  ##BoostOutline
│ │ │  Boost (Outline)
│ │ │  OpenBoardView
│ │ │ -Sun Jan 30 07:41:45 2022
│ │ │ +Sun Jan 30 16:33:06 2022
│ │ │  Build %s %s
│ │ │  http://openboardview.org
│ │ │  License info
│ │ │  Controls
│ │ │  KEYBOARD CONTROLS
│ │ │  Load file
│ │ │  Pan down
│ │ ├── readelf --wide --decompress --hex-dump=.rodata {}
│ │ │ @@ -572,15 +572,15 @@
│ │ │    0x004ff390 50696e73 00426f6f 73742028 50696e73 Pins.Boost (Pins
│ │ │    0x004ff3a0 29002323 426f6f73 74506172 74730042 ).##BoostParts.B
│ │ │    0x004ff3b0 6f6f7374 20285061 72747329 00232342 oost (Parts).##B
│ │ │    0x004ff3c0 6f6f7374 4f75746c 696e6500 426f6f73 oostOutline.Boos
│ │ │    0x004ff3d0 7420284f 75746c69 6e652900 446f6e65 t (Outline).Done
│ │ │    0x004ff3e0 0041626f 75740038 2e39352e 31004f70 .About.8.95.1.Op
│ │ │    0x004ff3f0 656e426f 61726456 69657700 53756e20 enBoardView.Sun 
│ │ │ -  0x004ff400 4a616e20 33302030 373a3431 3a343520 Jan 30 07:41:45 
│ │ │ +  0x004ff400 4a616e20 33302031 363a3333 3a303620 Jan 30 16:33:06 
│ │ │    0x004ff410 32303232 00427569 6c642025 73202573 2022.Build %s %s
│ │ │    0x004ff420 00687474 703a2f2f 6f70656e 626f6172 .http://openboar
│ │ │    0x004ff430 64766965 772e6f72 6700436c 6f736500 dview.org.Close.
│ │ │    0x004ff440 4c696365 6e736520 696e666f 00436f6e License info.Con
│ │ │    0x004ff450 74726f6c 73004b45 59424f41 52442043 trols.KEYBOARD C
│ │ │    0x004ff460 4f4e5452 4f4c5300 4c6f6164 2066696c ONTROLS.Load fil
│ │ │    0x004ff470 65005175 69740050 616e2075 70005061 e.Quit.Pan up.Pa
```

The `SOURCE_DATE_EPOCH` environment variable is set in the build environment but not honored (c.f.: https://reproducible-builds.org/docs/source-date-epoch/).

Thanks!

Apteryks, 2022-01-31:
One easy workaround is to substitute __TIMESTAMP__ with __DATE__, which *does* supports SOURCE_EPOCH_DATE.

piernov, 2022-02-06:
`__TIMESTAMP__` is indeed used both in `src/openboardview/main_opengl.cpp` and `src/openboardview/BoardView.cpp`.

Reproducible builds are not a focus in OBV development and it doesn't sound like this is an OBV-specific issue, however it is possible to replace `__TIMESTAMP__` with `__DATE__ " " __TIME__` if it fixes the issue I guess.

Apteryks, 2022-02-07:
__TIMESTAMP__ should have been made to fallback to something deterministic in the presence of SOURCE_EPOCH_DATE a long time ago but the patch submitted for doing so seems to have fallen into the cracks. Note that __TIMESTAMP__ is not standard though (but it's implemented by both GCC and Clang); so it may still be better to use `__DATE__ " " __TIME__`.

commit 8e1a12a6 (2022-03-20): Replace __TIMESTAMP__ with __DATE__ " " __TIME__ (closes #229)
closed 2022-03-20 by piernov