OpenBoardView/OpenBoardView #230: Error Opening file

tgolightly, 2022-02-23:
Hello,

I recently downloaded your software and tried to open a file downloaded from net-fixdotcom.  It displayed a terse message that there was an error open <filename went here>.  This appears to be similar to issue 56.  However, that user appears to be using the windows version.  I am using the latest linux release that was cloned from your git project repository.  I previously tried the .deb pkg and received the same popup window.  I have attached one of the 3 files that failed to load.  I really hope to be able to use your software, to help that small non-profit I am employed by to fix machines that might otherwise go to scrap.

Sincerely,
Terry Golightly
PC Technician
ComputerReach.org
[compresseddotbrd.tar.gz](https://github.com/OpenBoardView/OpenBoardView/files/8127124/compresseddotbrd.tar.gz)
y

piernov, 2022-02-23:
Allegro not supported. #62 #120 #126 #194 #225

tgolightly, 2022-02-25:
Hi Piernov,

Are  you aware of any reliable sources for .brd file that is actually a file that can be opened by your software or will this be hit or miss?

T

piernov, 2022-02-25:
File formats depend on what tools the manufacturer used for production/testing.
Several different formats use `.brd` extension so it doesn't mean a lot. 
Allegro files always contain `allvxx` (xx being the version it was created with) somewhere in the header.
Unfortunately OBV supports Landrex TestLink and TOPTEST BoardViewR4 formats only.

closed 2022-03-20 by piernov