OpenBoardView/OpenBoardView #234: Samsung .CAD files display artifacts

jcdutton, 2022-03-24:
Hi,

I have a Samsung .CAD file, and when using the latest openboardview from the git and compiling it. It nicely displays the board. Thank you for supporting this file format.
The problem is, that when I compare the display to the actual PCB board, the openboardview seems to have components, while mostly located in the right location, sometimes are displayed on the wrong side of the board than the PCB actually is.
For example, the power socket connector is shown on the wrong side of the board in openboardview.
Is this a known problem?
Can I help at all?
Where in the code is the .CAD file parsed, and the decision, on which side of the board it should be displayed on, taken?
Also, is the file format documented anywhere, so that I can compare that to the contents of the file, and determine if the original file has them wrong, or whether the problem is in openboardview?

piernov, 2022-03-24:
A sample file and pictures of the problem would be useful.

Samsung CAD parsing is done by `src/openboardview/FileFormats/CADFile.cpp`: https://github.com/OpenBoardView/OpenBoardView/blob/eed29c393f96dc32d42b1c97c56a6b7cc113826c/src/openboardview/FileFormats/CADFile.cpp
Decoding the part side is at line 93-96: https://github.com/OpenBoardView/OpenBoardView/blob/eed29c393f96dc32d42b1c97c56a6b7cc113826c/src/openboardview/FileFormats/CADFile.cpp#L93

I don't know about the format specifications. @skvatboy implemented the Samsung CAD format support, maybe they can help.

jcdutton, 2022-03-24:
Thank you for the pointers. I will look at the .cad file for the  1 and 2  and see if the original file is correct or not.
If the file is correct, and the view is wrong, I will create a patch to fix it.