OpenBoardView/OpenBoardView #239: Crashes Apple Silicon (arm64)

Bobronium, 2022-05-24:
Hello! I've tried to build the tool on MBP14 with M1 Pro. Build finished successfully and produced a .dmg, however .app inside crashed upon launching. Any ideas how it can be fixed? 

Also it would be great to have universal binary available for both architechtures.

###### inb4: installing rosetta is a last resort option, I'd want to avoid it if possible.

<details>
<summary>$ ./build.sh output</summary>

```py
$ ./build.sh
Compiling OpenBoardView using 10 threads (release build)
Extra flags passed to CMake: -DCMAKE_INSTALL_PREFIX=
-- The C compiler identification is AppleClang 13.1.6.13160021
-- The CXX compiler identification is AppleClang 13.1.6.13160021
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
OpenBoardView release: R821 /tags/9.0.2
-- Found PkgConfig: /opt/homebrew/bin/pkg-config (found version "0.29.2")
-- Found ZLIB: /Library/Developer/CommandLineTools/SDKs/MacOSX12.3.sdk/usr/lib/libz.tbd (found version "1.2.11")
-- Found SQLite3: /Library/Developer/CommandLineTools/SDKs/MacOSX12.3.sdk/usr/include (found version "3.37.0")
-- Found Python: /Users/bobronium/.pyenv/shims/python3.10 (found version "3.10.1") found components: Interpreter
-- Checking for std::filesystem
-- Checking for std::filesystem - not available, using ghc::filesystem
-- Found OpenGL: /Library/Developer/CommandLineTools/SDKs/MacOSX12.3.sdk/System/Library/Frameworks/OpenGL.framework
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/bobronium/dev/tools/OpenBoardView/release_build
[  2%] Generating GLAD
[05/24/2022 16:00:17][INFO	][glad   	]: downloading latest 'gl' specification
[05/24/2022 16:00:17][INFO	][glad.opener	]: opening: 'https://raw.githubusercontent.com/KhronosGroup/OpenGL-Registry/master/xml/gl.xml'
[05/24/2022 16:00:20][INFO	][glad   	]: generating 'gl' bindings
[05/24/2022 16:00:20][INFO	][glad.opener	]: saving: 'https://raw.githubusercontent.com/KhronosGroup/EGL-Registry/master/api/KHR/khrplatform.h' to '/Users/bobronium/dev/tools/OpenBoardView/release_build/src/glad/include/KHR/khrplatform.h'
[05/24/2022 16:00:20][INFO	][glad   	]: generating 'gl' bindings - done
[  2%] Built target glad-generate-files
[  4%] Building C object src/glad/CMakeFiles/glad.dir/src/glad.c.o
[  6%] Linking C static library libglad.a
[  8%] Built target glad
[ 10%] Building CXX object src/CMakeFiles/imgui.dir/imgui/imgui_draw.cpp.o
[ 14%] Building CXX object src/CMakeFiles/imgui.dir/imgui/misc/cpp/imgui_stdlib.cpp.o
[ 14%] Building CXX object src/CMakeFiles/imgui.dir/imgui/imgui.cpp.o
[ 16%] Building CXX object src/CMakeFiles/imgui.dir/imgui/imgui_widgets.cpp.o
[ 18%] Building CXX object src/CMakeFiles/imgui.dir/imgui/backends/imgui_impl_opengl2.cpp.o
[ 20%] Building CXX object src/CMakeFiles/imgui.dir/imgui/imgui_tables.cpp.o
[ 22%] Building CXX object src/CMakeFiles/imgui.dir/imgui/backends/imgui_impl_opengl3.cpp.o
[ 24%] Building CXX object src/CMakeFiles/imgui.dir/imgui/backends/imgui_impl_sdl.cpp.o
[ 26%] Linking CXX static library libimgui.a
[ 26%] Built target imgui
[ 28%] Building CXX object src/openboardview/CMakeFiles/openboardview.dir/annotations.cpp.o
[ 30%] Building CXX object src/openboardview/CMakeFiles/openboardview.dir/history.cpp.o
[ 34%] Building CXX object src/openboardview/CMakeFiles/openboardview.dir/BoardView.cpp.o
[ 34%] Building CXX object src/openboardview/CMakeFiles/openboardview.dir/utils.cpp.o
[ 36%] Building CXX object src/openboardview/CMakeFiles/openboardview.dir/confparse.cpp.o
[ 38%] Building CXX object src/openboardview/CMakeFiles/openboardview.dir/BRDBoard.cpp.o
[ 40%] Building CXX object src/openboardview/CMakeFiles/openboardview.dir/FileFormats/ASCFile.cpp.o
[ 42%] Building CXX object src/openboardview/CMakeFiles/openboardview.dir/vectorhulls.cpp.o
[ 44%] Building CXX object src/openboardview/CMakeFiles/openboardview.dir/FileFormats/ADFile.cpp.o
[ 46%] Building CXX object src/openboardview/CMakeFiles/openboardview.dir/FileFormats/BDVFile.cpp.o
[ 48%] Building CXX object src/openboardview/CMakeFiles/openboardview.dir/FileFormats/BRD2File.cpp.o
[ 51%] Building CXX object src/openboardview/CMakeFiles/openboardview.dir/FileFormats/BRDFile.cpp.o
[ 53%] Building CXX object src/openboardview/CMakeFiles/openboardview.dir/FileFormats/BVRFile.cpp.o
In file included from /Users/bobronium/dev/tools/OpenBoardView/src/openboardview/BRDBoard.cpp:1:
[... 399 lines cut ...]
      },
      "deploymentId" : 240000128
    }
  ],
  "experiments" : [

  ]
}
}

Model: MacBookPro18,3, BootROM 7459.121.3, proc 10:8:2 processors, 32 GB, SMC 
Graphics: Apple M1 Pro, Apple M1 Pro, Built-In
Display: Color LCD, 3024 x 1964 Retina, Main, MirrorOff, Online
Memory Module: LPDDR5
AirPort: Wi-Fi, wl0: Apr  6 2022 05:55:54 version 20.90.45.0.8.7.118 FWID 01-e7138ff2
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB31Bus
USB Device: USB31Bus
USB Device: USB31Bus
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.
```
</details>

piernov, 2022-05-24:
You'll have to sign the ARM64 binary for it to run on M1 Macs.
I don't have any M1 Mac (or even any recent Mac for that matter) so I can't provide support for it.

Bobronium, 2022-05-24:
Could you give some hints/search queries/instructions on how to sign ARM64 binary? My initial search didn't yield anything useful.

Out of curiosity: how other tools are able to build universal binaries in CI? GitHub Actions

piernov, 2022-05-24:
You'll most likely need an Apple developer account to sign with your own certificate. You can either generate an XCode project with CMake and link your developer account inside the XCode project, or use the `codesign` command line tool to sign an existing binary.

I don't know how other projects do it, I haven't looked into it, but either way, I wouldn't be able to test the app in a real environment without the required hardware. Previously we used Travis as a CI solution and it took care of building the macOS app as well, but since it became a mess I just build the app on my MacBook Pro 2007.

Bobronium, 2022-05-24:
Yay, finally got it to work. Thanks for help!

Had to:
```
codesign --force --deep --sign - /Applications/openboardview.app
```

[image]

piernov, 2022-05-29:
@cbusillo built a universal binary of the development release from an M1 machine and signed it with their Apple Developer certificate so it should run natively on both x86_64 and M1.
https://github.com/OpenBoardView/OpenBoardView/releases/download/9.90.0/OpenBoardView-9.90.0-Darwin.dmg

rudragthite, 2023-01-31:
@Bobronium , Thank you for the solution.

VladErc, 2023-08-29:
Neither the provided binary nor the command added by @Bobronium seem to work. Maybe it's due to MacOS updates maybe not but any help would be appreciated. I'm on an M2 Pro Mac Mini with the latest MacOS version (Ventura 13.5).

Tanpetor, 2024-05-12:
> Neither the provided binary nor the command added by @Bobronium seem to work. Maybe it's due to MacOS updates maybe not but any help would be appreciated. I'm on an M2 Pro Mac Mini with the latest MacOS version (Ventura 13.5).
> 
> L.E: Nevermind, I just had to go in the system settings -> security and allow the provided binary to open.

Can you tell please how to do it?

Tanpetor, 2024-05-12:
> Yay, finally got it to work. Thanks for help!
> 
> Had to:
> 
> ```
> codesign --force --deep --sign - /Applications/openboardview.app
> ```
> 
> [image]

What should I do in this situation

(base) tanpeter@MacBook-Pro ~ % codesign --force --deep --sign - /Applications/CLion.app 
    
/Applications/CLion.app: replacing existing signature
/Applications/CLion.app: Operation not permitted
In subcomponent: /Applications/CLion.app/Contents/Helpers/CLion.app