OpenBoardView/OpenBoardView #242: BVR is not working on Alpha release 9.0.3

quikfix94, 2022-05-29:
Hello, i am trying to open a BVR file specifically named "820-02016.bvr" and it is not opening, when trying to open it from the FILE -> OPEN menu, and selecting the BVR file it closes/crashes OpenBoardView without any error log/crash report.

the file line in the 820-02016.bvr file is "BVRAW_FORMAT_3". (as stated in previous reports that it requires it, and only supports that.

If you need me to attach the bvr file let me know. 

i am running Monteray 12.4.

piernov, 2022-05-30:
I cannot reproduce the problem, it works fine with OBV 9.0.3 on Monterey with this boardview file here. macOS should prompt you to give access to OBV to your documents when you try to open a file. You can try looking at the logs with the Console app, or try running OBV in a Terminal.

piernov, 2022-08-16:
Forgot about this one, make sure to use the 820-02016 V1.3 file as earlier revision were malformed and could cause a crash in some instances with the development version (*not* with 9.0.3). OBV master branch also has a fix that should prevent crashing with earlier revision of 820-02016 boardview.

closed 2022-08-16 by piernov