OpenBoardView/OpenBoardView #248: GenCAD board outline issue

swiftgeek, 2022-06-14:
(originally posted in https://github.com/OpenBoardView/OpenBoardView/pull/219#issuecomment-1147291206)
I have prepared a file with a board outline that still has issues loading properly
[board-outline.zip](https://github.com/OpenBoardView/OpenBoardView/files/8843636/board-outline.zip)

<details>
  <summary>Screenshots with expected behaviour/current state</summary>
  
 Expected board outline
[image]

Current OBV render:
[image]
  
</details>

piernov's response in pull request adding GenCAD support: https://github.com/OpenBoardView/OpenBoardView/pull/219#issuecomment-1147744835

inflex, 2022-06-15:
I suspect the problem is that the GenCAD outline usually is segment based, not a sequence of poly points. 

You've got two options that I can think of immediately;

1) Add segment-based outline + fill drawing versions of the existing outline/fill functions
2) Convert segment based outlines to polygon sequence ( note that internal holes will need to be handled either by "same point termination" or clockwise-anticlockwise path directions.

This is what it looks like in FlexBV ( which has the segment based rendering facility
[image]
 )

martonmiklos, 2022-06-16:
So here is an example gencad forming a circle contour with 100 radius around the 100,100 point:
https://gist.github.com/martonmiklos/27564ca2fc0ac7108c7fccc117e1413a

This renders this way:
[image]

commit 93ce4f07 (2022-08-25): Fix GenCAD outline parsing and use segmented mode (closes #248)
closed 2022-08-25 by piernov