OpenBoardView/OpenBoardView #253: Feature request: Flip board direction setting

swiftgeek, 2022-06-23:
Would be nice to have a setting to control whether board is flipped vertically or horizontally (currently it seems to follow rotation)

inflex, 2022-06-23:
That's one of those settings that no matter what you set it to ( as a user ) it's almost always going to be the other setting you want ;)

LabGoblin, 2024-10-16:
I would also like to see a 'flip' the way EDA software (like KiCad) does it. Or I guess it would be more of an 'x-ray' ? 

It would make following a trace on a board image 'through' the board way less confusing

rzsn, 2026-08-27:
I would appreciate if the FLIP would actually keep the viewport.

Current state: the board flips along its center (or min/max average), while viewport stays - leading to the need to move the image.
If I screenshot the two flipped sides, they do not overlap.

Desired: flip would happen at the position of the viewport (window) center.
If I screenshot the two flipped sides, they would perfectly overlap.

( full disclosure: I was using a TVW file from the fork, not sure if the particular file/format matters .. should not )