OpenBoardView/OpenBoardView #254: Component and copper sides are inverted

Guimli, 2022-07-18:
Hello
I converted a PCB from Kicad with KiCAD to Boardview exporter. And the boardview appears inverted in OpenBoardView. But not in other Boardview software.
[https://github.com/Guimli/Vectrex-Schematic/raw/main/LogicBoard%203GE/LogicBoard%203GE.brd](url)

Guimli, 2022-07-18:
[image]
[image]

piernov, 2022-08-21:
Boardview can be mirrored by pressing the `M` key or by clicking on "Mirror Board" in the "View" menu.

sabogalc, 2024-03-23:
Hey, this issue should have been fixed with [this PR from the KiCad to Boardview exporter](https://github.com/whitequark/kicad-boardview/pull/12). Please let me know if it works now!

Guimli, 2024-03-29:
Hello
I can confirm that an export with the latest version of KiCad to Boardview exporter fixes the problem.
Sorry to have opened the issue on OpenBoardView. It was a problem with export.
Thanks again for OpenBoardView.

sabogalc, 2024-03-29:
> Hello I can confirm that an export with the latest version of KiCad to Boardview exporter fixes the problem. Sorry to have opened the issue on OpenBoardView. It was a problem with export. Thanks again for OpenBoardView.

That's great to hear! Since it is resolved, can you close the ticket here please?

Guimli, 2024-03-30:
Yes

closed 2024-03-30 by Guimli