OpenBoardView/OpenBoardView #258: GenCAD NB8513_MB_V3.cad $BOARD and $DEVICES parsing error

piernov, 2022-08-21:
Attached file does not open with master branch.
`$BOARD` does not get parsed properly due to the `TEXT` entry not in the grammar, so just a matter of adding it I guess.
`$DEVICES` does not get parsed properly due to `DEVICE LRC  LBAT54CWT1` having multiple unquoted words I think. No idea what this is about.

@swiftgeek or @galkinvv if one of you want to take a look at it.

[NB8513_MB_V3.zip](https://github.com/OpenBoardView/OpenBoardView/files/9389578/NB8513_MB_V3.zip)

swiftgeek, 2022-08-21:
Looks like newer CAMCAD (likely one included with PADS) started doing something unexpected, or input file got somewhat corrupted on import/export to product this strange result

piernov, 2022-08-22:
Do not hesitate to post a new comment rather than editing your previous one if you have new information, we do not get notified of edits.

If you think this really is a malformed file we could simply edit it to fix it, but if there's a possibility there are other files out there with the same kind of issues it's best for the parser to handle these cases.
Also, in general it would be better for the parser to be more permissive but I assume it's not easy with a grammar-based parsing.

As for the actual parser code, I cannot provide any insight unfortunately. Maybe @galkinvv can share some ideas if he can spare some time to look at it.

Otherwise, if you think your changes are "good enough" to handle the various files we've seen until now we can merge that. If it breaks we can fix it in a future beta release (although it seems like nobody really test alpha or beta releases).

swiftgeek, 2022-08-22:
I would rather want IRC channel on libera or oftc working like before, if possible ^^

CAMCAD is probably the most used used exporter (though maybe an older version of it), so I will need to test later what exactly is to blame. GenCAD itself is pretty widely used though, so that divergence from spec is quite disturbing.

GenCAD support is pretty important to me and the last thing I would want is it being broken/regressing, especially when I'm not sure what exactly am I doing to the code.

galkinvv, 2022-08-22:
`GenCADFile::get_device_by_name` was added by @martonmiklos during creating of first WIP prototype. Maybe there were some plans for using it, but now gencad renders well without it, so I think deleting it is safe by now.

About regressions - the best idea I have now - have a set of dome different cad files to be able at least test manually.
Some such files has "shady" source, so I'm not sure that attaching them to the projects github is a good idea.

I've created a google shared folder with them (and a subfolder with files having rendering problems).
I've uploaded there 10 files that I'm using for basic "visually opens fine" regression testing.

This link allows **uploads** by anyone - you can add new files. Also I plan to add there some kicad sample files from github discussions last months.

Note: since this link allow uploads - I protect it from bots by adding 6 symbols "NoBots". Manually edit the link before following.

https://drive.google.com/drive/folders/NoBots1XD3OSYpvj5znjMmG58cHdOut3f2ltg--

About discussions - really, I'm using only telegram. If it is ok for you we can publicly discuss things in https://t.me/OpenBoardViewUnofficial (I've just created it)

swiftgeek, 2022-08-22:
I have extremely bad luck with proprietary networks, got banned there without sending a single message ;D

[Also this](https://xkcd.com/1782/)

swiftgeek, 2022-08-22:
Yeah so even CAMCAD doesn't see valid data there (like actual board outline) so I would mark this particular sample as partially corrupted due to user error.

Normally with CAMCAD I would import whatever (eg. extracta export), remap `$$DEVICE$$` to `PARTNUMBER` with "Keyword Map Editor", export to GenCAD first time, close and reimport, and then export with just the board layer enabled - which trims out entire artwork and leaves only stuff required for stencils and OBV.

Or import to KiCAD and then process with https://github.com/swiftgeek/pcbtools/blob/main/gencad/gencad_strip_art.sh to strip artwork.

piernov, 2022-08-23:
A bit off-topic but about the IRC channel, basically there hasn't been any activity in years on #openboardview@freenode (except a few random people coming in asking for FZ key or schematics) and I also gradually left IRC due to no activity in the other channels I was in (my client is still running though); the freenode nonsense was enough to completely give up on it (hence why I cleaned up all references to the IRC channel).
However if there is a use for it I guess we can create a new channel on libera.chat or OFTC (no idea which one makes more sense) . 

About attaching files to issue, let's say that you "should not" attach copyrighted/confidential content, but if it happens, well, it can easily be removed later on request. The thing that matters is that the git history does not get polluted with such data (hence why we still do not include the FZ key as it could be litigious).

swiftgeek, 2022-08-23:
libsdl and kicad community channel are on libera so I guess libera makes most sense

swiftgeek, 2022-08-23:
Ok so this issue/quirk is resolved with #259 merged

closed 2022-08-23 by piernov