OpenBoardView/OpenBoardView #266: GENCAD 1.4 - Unrecognised file format

dfear, 2022-11-28:
Dell latitude 3490 boardview file in  download in ".cad" format.

Opening the file results in: Unrecognised file format.

The header of the file is: 
$HEADER
GENCAD 1.4
USER "MENTOR GRAPHICS - TRANSLATOR CAMCAD Professional 4.14.312.6 - Licensed"
$ENDHEADER

[Dell LATITUDE 3490 3590 - LA-F115P - LA-F116P Boardview (.CAD).cad.zip](https://github.com/OpenBoardView/OpenBoardView/files/10105242/Dell.LATITUDE.3490.3590.-.LA-F115P.-.LA-F116P.Boardview.CAD.cad.zip)

piernov, 2022-11-28:
GenCAD support is not available in a release yet, need to build OBV from master branch.

piernov, 2023-02-13:
https://github.com/OpenBoardView/OpenBoardView/releases/tag/9.95.0

closed 2023-02-13 by piernov