OpenBoardView/OpenBoardView #267: Feature request (not really) don't require an internet connection for building from source.

heldiho53, 2023-01-10:
Build systems like the open build service or koji don't have access to the internet while building. Some of the python packages try to call an URL. Obviously without internet building will fail. I don't think it's good practice to require an internet connection to build the code from source anyways. So please consider making these changes.

MattSturgeon, 2023-01-10:
> I don't think it's good practice to require an internet connection to build the code from source

Agreed :+1: 

> Some of the python packages try to call an URL

Have you identified specific dependencies we should try to remove (or improve upstream)?

piernov, 2023-01-10:
Some of the dependencies supplied as git submodules can be installed system-wise, for the ones that cannot, the git repository can be cloned with `--recursive` to fetch them all at once before the build.

However, the OpenGL loader glad requires an Internet connection regardless to download the OpenGL spec files. I don't know if it's possible to fetch them before running the build. You can check glad's documentation to see if it's possible, or talk with glad's developers.
As a side note, I don't think it's a good idea to include generated files in our repository either.

heldiho53, 2023-01-10:
> > I don't think it's good practice to require an internet connection to build the code from source
> 
> Agreed +1
> 
> Have you identified specific dependencies we should try to remove (or improve upstream)?
Sorry I haven't. Unfortunately I don't have the time either at the moment

> 

> Some of the dependencies supplied as git submodules can be installed system-wise, for the ones that cannot, the git repository can be cloned with `--recursive` to fetch them all at once before the build.

Well the recursive cloning isn't the issue, that I've done beforehand. None of the releases actually builds without the submodules

MattSturgeon, 2023-01-10:
It's been a while since I looked, but it should be that all dependencies are either got submodules and/or system libraries. If building directly with cmake there shouldn't be anything to download, ideally.

However there is also a helper build.sh script that may automate some of the dependency fetching. Again, a while since I looked so I may be wrong on some of the details...

I don't know how extensively we've documented the ability to build without using build.sh either...

piernov, 2023-01-10:
`build.sh` doesn't do anything fancy, basically it just calls `cmake` and `make`, makes it easier for people who have no idea what CMake is to build the project. It's the CMake build scripts that'll init/update the git submodules as needed but you can do that beforehand if you want.
As I said though, it's the OpenGL loader glad that necessarily downloads a file during the build and I don't think it's something we have control over.

commit bc389b0a (2023-01-15): glad reproducible build (closes #267)

Reproducible builds use bundled OpenGL specs instead of downloading during build
closed 2023-01-15 by piernov