OpenBoardView/OpenBoardView #273: Release and pre-release versioning

k3a, 2023-02-26:
I am a maintainer of Nix [derivation](https://github.com/NixOS/nixpkgs/pull/217922/files) (build recipe) for OBV.

I can see that this repo has various pre-releases (often in the X.9Y.Z format) but the latest one is not marked as pre-release, even though it is 9.95.0.

1. Should pre-releases be officially distributed or should they be skipped? I suppose skipped.
2. Should the latest 9.95.0 be distributed or skipped? I suppose skipped.

Thanks for clarification!

piernov, 2023-02-27:
Release management is indeed a bit of a mess and there is no planned schedule.
I removed the pre-release marker from the 9.95.0 beta release so that it is shown as the latest release. Otherwise, people never see it and never try it, and I don't know how long will it be until the next stable (10.0.0) release.
There are a few useful bug fixes and additions compared to 9.0.3 and hopefully not many more broken things.
It is still a "beta" release but honestly the "stable" releases aren't really "stable" either, so it is up to you (and how it is usually done for this distribution) to ship it or not.

x.90.x are alphas, x.95.x are betas, this is done because some package managers can be quite restrictive on the allowed string for version.

k3a, 2023-02-27:
Thanks, I think it is fine, then! ;)

I would expect a pre-release to be on something that is not yet usable and is only useful for brave "testing" users.
For example, a rewrite that supports half of the formats supported in the previous stable (with work in progress to add the remaining previously-supported formats) or a rewrite from OpenGL to Vulkan that occasionally crashes for some users. But it's up to you.

I think that Nix auto-update script ignores Github pre-releases by default, btw.

closed 2023-02-27 by k3a