OpenBoardView/OpenBoardView #282: GNU/Linux 9.95.0 crashes when attempting to load a file

Apteryks, 2023-03-31:
Hello,

I've updated the Guix package `openboardview`, but when I try to load a file, I get:

$ /gnu/store/3855sc2pmg7qrmsd0bi1wwr69fnxwf9b-openboardview-9.95.0/bin/openboardview 
INFO: Initializing ImGuiRendererSDLGL3
INFO: 
-------------------------------------------------------------
GL Vendor     : nouveau
GL GLRenderer : NV50
GL Version    : 3.3 (Core Profile) Mesa 21.3.8
GLSL Version  : 3.30
-------------------------------------------------------------
ERROR: OpenGL3 renderer not supported on nouveau NV50
INFO: Initializing ImGuiRendererSDLGL1
INFO: 
-------------------------------------------------------------
GL Vendor     : nouveau
GL GLRenderer : NV50
GL Version    : 3.3 (Compatibility Profile) Mesa 21.3.8
GLSL Version  : 3.30
-------------------------------------------------------------
maxim@hurd ~/src/guix-staging [env]$ LC_ALL=C /gnu/store/3855sc2pmg7qrmsd0bi1wwr69fnxwf9b-openboardview-9.95.0/bin/openboardview 
INFO: Initializing ImGuiRendererSDLGL3
INFO: 
-------------------------------------------------------------
GL Vendor     : nouveau
GL GLRenderer : NV50
GL Version    : 3.3 (Core Profile) Mesa 21.3.8
GLSL Version  : 3.30
-------------------------------------------------------------
ERROR: OpenGL3 renderer not supported on nouveau NV50
INFO: Initializing ImGuiRendererSDLGL1
INFO: 
-------------------------------------------------------------
GL Vendor     : nouveau
GL GLRenderer : NV50
GL Version    : 3.3 (Compatibility Profile) Mesa 21.3.8
GLSL Version  : 3.30
-------------------------------------------------------------

(process:11341): GLib-GObject-CRITICAL **: 16:07:03.328: g_object_unref: assertion 'G_IS_OBJECT (object)' failed

(process:11341): GLib-GObject-CRITICAL **: 16:07:03.333: g_object_unref: assertion 'G_IS_OBJECT (object)' failed
ERROR: PDFBridgeEvince PDF file does not exist
openboardview: imgui.cpp:4660: void AddDrawListToDrawData(ImVector<ImDrawList*>*, ImDrawList*): Assertion `draw_list->_VtxCurrentIdx < (1 << 16) && "Too many vertices in ImDrawList using 16-bit indices. Read comment above"' failed.
```
Would you happen to have an idea of what this failure may be caused by?  The libraries used to build are:
```
fontconfig-minimal@2.13.94 glad@0.1.36 gtk+@3.24.30 imgui@1.88
+ orangeduck-mpc@0.9.0-0.7c910e9 pkg-config@0.29.2 python@3.9.9 sdl2@2.26.2 sqlite@3.36.0 stb-image@2.26
+ utf8-h@0.0.0-0.500d4ea zlib@1.2.11
```

Thanks!

piernov, 2023-03-31:
ImGui needs to be built with `-DImDrawIdx=unsigned int` as it is the case for the bundled one.

Apteryks, 2023-04-03:
Thanks, that fixed it!

closed 2023-04-03 by Apteryks